RobotLocomotion/drake #14235: Create a helper function to verify a mesh of a shape primitive that uses medial axis.

DamrongGuoy, 2020-10-21:
We plan to generate tetrahedral meshes of a number of shape primitives (e.g., Box, Capsule, Cylinder). It would be helpful to have a helper function for verification that a mesh conforms to the medial axis of a shape primitive. We can use it in our unit tests or verifying user-supplied meshes.

Most of the idea came from @joemasterjohn .

A possible function declaration is:

```C++
bool IsTetrahedronRespectingMa(
       const VolumeElement& tetrahedron,
       const VolumeMesh<double>& mesh,
       const Shape& shape,
       std::function<double(const Shape&, const Vector3d&)> distance_to_boundary,
       std::function<double(const Shape&, int, const Vector3d&)> distance_to_face_i,
       std::function<int(const Shape&)> num_faces,    
       const double tolerance)
```
A possible algorithm for ` IsTetrahedronRespectingMa()` is
```C++
for each vertex v:
  let f[v] bet the set of "closest faces" (up to some tolerance)
check that the intersection of all f[v] is one face
```
It shows that the vertices share a closest boundary face, and only one closest face (i.e. they are in the same block of the medial-axis subdivision but not all on the same face of the medial-axis).

It will be useful for #11906, #12128, and #13745.

SeanCurtis-TRI, 2021-03-22:
Changing priority to reflect actual action/scheduling.

jwnimmer-tri, 2024-03-20:
> Changing priority to reflect actual action/scheduling.

Indeed, I don't think we're adding more new hydro primitive shapes anymore, are we?  So "not planned" to add more test fixtures.

closed 2024-03-20 by jwnimmer-tri