RobotLocomotion/drake #14243: missing scipy dependency on mac

RussTedrake, 2020-10-23:
I don't love that we have that dependency, but it appears that we do, and that it's suddenly not being installed by our mac setup scripts.

```
packages/pydrake/systems/all.py", line 7, in <module>
    from .planar_scenegraph_visualizer import *
  File "/private/var/tmp/_bazel_runner/7b4dc6001ecdbeb1b5ef1e14edba8c37/sandbox/darwin-sandbox/257/execroot/manipulation/bazel-out/darwin-fastbuild/bin/exercises/pose/pose_estimation_icp.runfiles/drake/.lib/python3.8/site-packages/pydrake/systems/planar_scenegraph_visualizer.py", line 15, in <module>
    import scipy as sp
ModuleNotFoundError: No module named 'scipy'
```

(full output here: https://github.com/RussTedrake/manipulation/runs/1296900627?check_suite_focus=true)

The dependency appears to come in from our use of `spatial.ConvexHull` in `planar_scenegraph_visualizer`.    Perhaps #14197 will provide an alternative.

jwnimmer-tri, 2020-10-23:
The `scipy` dependency _is_ there in

https://github.com/RobotLocomotion/drake/blob/63691155d04cce67c3c333e609b4b2a6207dd2f8/setup/mac/binary_distribution/Brewfile#L25

It looks like homebrew has started into the python3.9 transition to some extent.  Perhaps some paths are wrong now?

Indeed, from https://formulae.brew.sh/formula/scipy it looks like scipy is 3.9-only now.  So we're back to the same shenanigans as #13097 where it's difficult to obtain a working system except through a narrow window of $PATH setting and dependency management.

jwnimmer-tri, 2020-10-23:
From a quick skim of https://github.com/Homebrew/homebrew-core/issues/62201, perhaps most of what we need is already in homebrew for python3.9, so the answer might be to switch Drake's pinned python on macOS to be 3.9 instead of 3.8?

jamiesnape, 2020-11-11:
Current list of failures:

* //:py/install_test
* //bindings/pydrake:py/all_each_import_test
* //bindings/pydrake:py/parse_models_test
* //bindings/pydrake/systems:py/monte_carlo_test
* //bindings/pydrake/systems:py/rendering_test
* //bindings/pydrake/systems:py/sensors_test
* //bindings/pydrake/visualization:py/plotting_test
* //examples/van_der_pol:plot_limit_cycle_test
* //tools/workspace/drake_visualizer:py/drake_visualizer_bazel_test

` //:py/install_test` and `//tools/workspace/drake_visualizer:py/drake_visualizer_bazel_test` probably just need another rebuild of VTK and Drake Visualizer (in progress, trying to avoid having to patch anything). A few of the others have all their tests passing, but bazel reporting failure, so something is unhappy in the teardown.

jamiesnape, 2020-11-18:
I think we may need to take a different tack and try to keep Python 3.8 support alive. I am still getting random crashes, and it seems director's dependencies do not like Python 3.9 either. The pip `numpy` appears to have issues related to bugs `Accelerate` that cause `matplotlib`, `pandas`, and `scipy` to fail, so we would need to build our own `numpy`, probably via Homebrew.

jwnimmer-tri, 2020-11-18:
For director, I think that its runtime does not need to be the same as the rest of our programs -- it doesn't use any binary modules from Drake, so it can stay back on the 3.8 runtime for a while, I think?  Or if that's too difficult, we can just immediately remove it from the macOS support matrix for the time being.  Or, maybe shipping it as a bottle is sufficient, and we can drop it from the workspace wrapping entirely?

For the other crashes, I don't quite understand the description -- I thought that we we using `numpy` from brew, not `pip`, so mention of "pip numpy" doesn't make sense.

https://github.com/RobotLocomotion/drake/blob/3ac311c7d798a42e77ffbe9793a9f12d793add8b/setup/mac/binary_distribution/Brewfile#L21

jamiesnape, 2020-11-18:
> For the other crashes, I don't quite understand the description -- I thought that we were using numpy from brew, not pip, so mention of "pip numpy" doesn't make sense.

We do, I was noting that the most obvious way of getting `numpy` for python 3.8 would not work. The crashes are not `numpy`-related as far as I can tell.

jwnimmer-tri, 2020-11-18:
> For director, ...

I should clarify the thought process here, rather than chasing through my specific suggestions.  I would like to scope fixing director on macOS as minimally as possible, so that we can get a release out with a working pydrake on current macOS.  If there is no quick fix, then we drop it now and try to resurrect it later.  If there is an easy fix like making a 3.8 bottle, we can do that too.

Bottom line, I'd like to invest more time in pydrake and less time in director.

jamiesnape, 2020-11-18:
Yes, director is not my main concern, it just complicates life. If we can disable it on Mac for now, it would be good.

RussTedrake, 2020-11-19:
Disabling director on mac for now seems very reasonable to me.  We have higher priority issues to address.

jwnimmer-tri, 2021-02-18:
This is all done.  The one follow-up is #14650.

closed 2021-02-18 by jwnimmer-tri