RobotLocomotion/drake #14250: pip3 installing matplotlib on ubuntu 18.04 breaks bazel tests.

RussTedrake, 2020-10-24:
the ubuntu 18.04 package manager serves up matplotlib 2.1.1 (very old)
it's very easy to accidentally install matplotlib via pip, e.g. by `pip3 install mpld3` (as we do in the manipulation repo), because it wants a newer version of matplotlib.

that should be fine.  But for some reason I can't quite figure out, it breaks bazel.  specifically, python bazel tests find the new (pip installed) version of matplotlib, but do *not* find the pip installed version of it's dependency `mpl_toolkits`.  So I end up with 
```
  File "/home/russt/.cache/bazel/_bazel_russt/a0bdb2099cb05916281dea472bfce61b/sandbox/linux-sandbox/3115/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/planar_scenegraph_visualizer_test.runfiles/drake/bindings/pydrake/systems/test/planar_scenegraph_visualizer_test.py", line 19, in <module>
    from pydrake.systems.planar_scenegraph_visualizer import (
  File "/home/russt/.cache/bazel/_bazel_russt/a0bdb2099cb05916281dea472bfce61b/sandbox/linux-sandbox/3115/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/planar_scenegraph_visualizer_test.runfiles/drake/bindings/pydrake/systems/planar_scenegraph_visualizer.py", line 7, in <module>
    import matplotlib.pyplot as plt
  File "/home/russt/.local/lib/python3.6/site-packages/matplotlib/pyplot.py", line 43, in <module>
    from matplotlib.figure import Figure, figaspect
  File "/home/russt/.local/lib/python3.6/site-packages/matplotlib/figure.py", line 18, in <module>
    from matplotlib import docstring, projections
  File "/home/russt/.local/lib/python3.6/site-packages/matplotlib/projections/__init__.py", line 4, in <module>
    from mpl_toolkits.mplot3d import Axes3D
  File "/usr/lib/python3/dist-packages/mpl_toolkits/mplot3d/__init__.py", line 6, in <module>
    from .axes3d import Axes3D
  File "/usr/lib/python3/dist-packages/mpl_toolkits/mplot3d/axes3d.py", line 32, in <module>
    from matplotlib.cbook import _backports
ImportError: cannot import name '_backports'
```

to reproduce, it should be sufficient to `pip3 install matplotlib`.  `pip3 uninstall matplotlib` resolves the issue for bazel (but is not the right solution).

My behavior from the command line is
```
russt@Puget-179850-01:~/drake$ python3 -c "from importlib.util import find_spec; print(find_spec('matplotlib'))"
ModuleSpec(name='matplotlib', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f35cb046b70>, origin='/home/russt/.local/lib/python3.6/site-packages/matplotlib/__init__.py', submodule_search_locations=['/home/russt/.local/lib/python3.6/site-packages/matplotlib'])
russt@Puget-179850-01:~/drake$ python3 -c "from importlib.util import find_spec; print(find_spec('mpl_toolkits'))"
ModuleSpec(name='mpl_toolkits', loader=None, origin='namespace', submodule_search_locations=_NamespacePath(['/home/russt/.local/lib/python3.6/site-packages/mpl_toolkits', '/usr/lib/python3/dist-packages/mpl_toolkits']))
russt@Puget-179850-01:~/drake$ python3 -c "import sys; print(sys.path)"
['', '/home/russt/underactuated', '/home/russt/manipulation', '/home/russt/drake-install/lib/python3.6/site-packages', '/home/russt/drake', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/home/russt/.local/lib/python3.6/site-packages', '/usr/local/lib/python3.6/dist-packages', '/usr/lib/python3/dist-packages']
```

but running this code
```
from importlib.util import find_spec
print(find_spec('matplotlib'))
print(find_spec('mpl_toolkits'))
import sys
print(sys.path)
```
from a bazel python test results in 
```
ModuleSpec(name='matplotlib', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fa7f99fcf60>, origin='/home/russt/.local/lib/python3.6/site-packages/matplotlib/__init__.py', submodule_search_locations=['/home/russt/.local/lib/python3.6/site-packages/matplotlib'])
ModuleSpec(name='mpl_toolkits', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fa8896607f0>, origin='/usr/lib/python3/dist-packages/mpl_toolkits/__init__.py', submodule_search_locations=['/usr/lib/python3/dist-packages/mpl_toolkits'])
['/home/russt/drake/common/test_utilities', '/home/russt/.cache/bazel/_bazel_russt/a0bdb2099cb05916281dea472bfce61b/sandbox/linux-sandbox/3117/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/planar_scenegraph_visualizer_test.runfiles', '/home/russt/.cache/bazel/_bazel_russt/a0bdb2099cb05916281dea472bfce61b/sandbox/linux-sandbox/3117/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/planar_scenegraph_visualizer_test.runfiles/drake/bindings', '/home/russt/.cache/bazel/_bazel_russt/a0bdb2099cb05916281dea472bfce61b/sandbox/linux-sandbox/3117/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/planar_scenegraph_visualizer_test.runfiles/lcmtypes_bot2_core/lcmtypes', '/home/russt/.cache/bazel/_bazel_russt/a0bdb2099cb05916281dea472bfce61b/sandbox/linux-sandbox/3117/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/planar_scenegraph_visualizer_test.runfiles/lcmtypes_bot2_core', '/home/russt/.cache/bazel/_bazel_russt/a0bdb2099cb05916281dea472bfce61b/sandbox/linux-sandbox/3117/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/planar_scenegraph_visualizer_test.runfiles/lcmtypes_robotlocomotion/lcmtypes', '/home/russt/.cache/bazel/_bazel_russt/a0bdb2099cb05916281dea472bfce61b/sandbox/linux-sandbox/3117/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/planar_scenegraph_visualizer_test.runfiles/lcmtypes_robotlocomotion', '/home/russt/.cache/bazel/_bazel_russt/a0bdb2099cb05916281dea472bfce61b/sandbox/linux-sandbox/3117/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/planar_scenegraph_visualizer_test.runfiles/ignition_math', '/home/russt/.cache/bazel/_bazel_russt/a0bdb2099cb05916281dea472bfce61b/sandbox/linux-sandbox/3117/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/planar_scenegraph_visualizer_test.runfiles/lcm', '/home/russt/.cache/bazel/_bazel_russt/a0bdb2099cb05916281dea472bfce61b/sandbox/linux-sandbox/3117/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/planar_scenegraph_visualizer_test.runfiles/drake', '/home/russt/.cache/bazel/_bazel_russt/a0bdb2099cb05916281dea472bfce61b/sandbox/linux-sandbox/3117/execroot/drake/bazel-out/k8-opt/bin/bindings/pydrake/systems/py/planar_scenegraph_visualizer_test.runfiles/spdlog', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/usr/local/lib/python3.6/dist-packages', '/home/russt/.local/lib/python3.6/site-packages', '/usr/lib/python3/dist-packages']
```
and then the breakage.

jwnimmer-tri, 2020-10-26:
I haven't tried to check the output above in detail, but on Ubuntu our goal is to ignore `~/.local` entirely via these lines:

https://github.com/RobotLocomotion/drake/blob/6bd5f8c25cdcbb4049e57ec0f49f6e2c136d864a/tools/ubuntu-bionic.bazelrc#L11-L15

I wonder why they are not working?  Can you check that your prereqs are set up correctly?
```console
jwnimmer@call-cps:~/jwnimmer-tri/drake$ cat gen/environment.bazelrc 
import %workspace%/tools/ubuntu-bionic.bazelrc
```

RussTedrake, 2020-10-26:
That much looks ok:
```
russt@Puget-179850-01:~/drake$ cat gen/environment.bazelrc 
import %workspace%/tools/ubuntu-bionic.bazelrc
```

jamiesnape, 2020-10-27:
My initial reaction was that a matplotlib-installed *.pth file was involved. I will look into that tomorrow.

RussTedrake, 2020-10-28:
The same(?) issue is biting me in my manipulation repo, but in a different way.  This time I have not installed anything to my user pip.  bazel is pulling down a newer version of pip from the requirements.txt, and is (again) finding the bazel-installed new version of matlab, but finding the system installed old version of mpl_toolkit.

```
usr/lib/python3.6/runpy.py:125: RuntimeWarning: 'meshcat.servers.zmqserver' found in sys.modules after import of package 'meshcat.servers', but prior to execution of 'meshcat.servers.zmqserver'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
Matplotlib created a temporary config/cache directory at /tmp/matplotlib-7x_ybxpc because the default path (/home/russt/.config/matplotlib) is not a writable directory; it is highly recommended to set the MPLCONFIGDIR environment variable to a writable directory, in particular to speed up the import of Matplotlib and to better support multiprocessing.
Traceback (most recent call last):
  File "/home/russt/.cache/bazel/_bazel_russt/7bcbcd19e064ca0197e4cfaf8f9df35e/sandbox/linux-sandbox/98/execroot/manipulation/bazel-out/k8-fastbuild/bin/trajectories.runfiles/manipulation/trajectories.ipynb.py", line 43, in <module>
    import pydrake.all
  File "/home/russt/.cache/bazel/_bazel_russt/7bcbcd19e064ca0197e4cfaf8f9df35e/sandbox/linux-sandbox/98/execroot/manipulation/bazel-out/k8-fastbuild/bin/trajectories.runfiles/drake/.lib/python3.6/site-packages/pydrake/all.py", line 46, in <module>
    from .systems.all import *
  File "/home/russt/.cache/bazel/_bazel_russt/7bcbcd19e064ca0197e4cfaf8f9df35e/sandbox/linux-sandbox/98/execroot/manipulation/bazel-out/k8-fastbuild/bin/trajectories.runfiles/drake/.lib/python3.6/site-packages/pydrake/systems/all.py", line 7, in <module>
    from .planar_scenegraph_visualizer import *
  File "/home/russt/.cache/bazel/_bazel_russt/7bcbcd19e064ca0197e4cfaf8f9df35e/sandbox/linux-sandbox/98/execroot/manipulation/bazel-out/k8-fastbuild/bin/trajectories.runfiles/drake/.lib/python3.6/site-packages/pydrake/systems/planar_scenegraph_visualizer.py", line 7, in <module>
    import matplotlib.pyplot as plt
  File "/home/russt/.cache/bazel/_bazel_russt/7bcbcd19e064ca0197e4cfaf8f9df35e/sandbox/linux-sandbox/98/execroot/manipulation/bazel-out/k8-fastbuild/bin/trajectories.runfiles/pip_pypi__matplotlib_3_3_2/matplotlib/pyplot.py", line 43, in <module>
    from matplotlib.figure import Figure, figaspect
  File "/home/russt/.cache/bazel/_bazel_russt/7bcbcd19e064ca0197e4cfaf8f9df35e/sandbox/linux-sandbox/98/execroot/manipulation/bazel-out/k8-fastbuild/bin/trajectories.runfiles/pip_pypi__matplotlib_3_3_2/matplotlib/figure.py", line 18, in <module>
    from matplotlib import docstring, projections
  File "/home/russt/.cache/bazel/_bazel_russt/7bcbcd19e064ca0197e4cfaf8f9df35e/sandbox/linux-sandbox/98/execroot/manipulation/bazel-out/k8-fastbuild/bin/trajectories.runfiles/pip_pypi__matplotlib_3_3_2/matplotlib/projections/__init__.py", line 4, in <module>
    from mpl_toolkits.mplot3d import Axes3D
  File "/usr/lib/python3/dist-packages/mpl_toolkits/mplot3d/__init__.py", line 6, in <module>
    from .axes3d import Axes3D
  File "/usr/lib/python3/dist-packages/mpl_toolkits/mplot3d/axes3d.py", line 32, in <module>
    from matplotlib.cbook import _backports
ImportError: cannot import name '_backports'
```

Currently the only way I can successfully run tests on this repo on ubuntu is by *installing* matplotlib via my user pip (the opposite of the uninstall recipe I have for drake).  I've also confirmed that adding the lines @jwnimmer-tri referred to to my bazelrc makes the behavior identical to drake (unfortunately, this is not successful here, since I actually *need* the newer matplotlib).  

So, for now, I have to install user pip matplotlib every time I want to run tests on manipulation, and uninstall it if i want to run tests on drake.  (yes, I might finally adopt virtualenv).  I'll leave the priority as medium for the world.  But it's a high priority for me to resolve this.

jwnimmer-tri, 2020-10-28:
For the original issue (Drake tests failing), I am unable to repro, at least not with a simple recipe.

My test case is `bazel test //bindings/pydrake/systems:py/planar_scenegraph_visualizer_test --nocache_test_results --test_output=streamed`.

Even after `pip3 install matplotlib`, it still passes.
Even then doing `python3 -c 'import matplotlib.pyplot'`, it still passes.

I suspect there must be some _additional_ kind of use of matplotlib that leaves residue somewhere that leaks into the bazel sandbox.  But I'm not sure how to make it happen.

Once we do have a repro, then we can try to narrow down where the local path infection happens, and then work on defending against it.

https://matplotlib.org/faq/troubleshooting_faq.html has some tips.  Maybe `$HOME.config/matplotlib` is where the leak comes from?

jwnimmer-tri, 2020-10-28:
If it _is_ the config folder leaking, then maybe setting `build --test_env=MPLCONFIGDIR=` or `build --test_env=MPLCONFIGDIR=/does/not/exist` in `tools/bazel.rc` would resolve it.

RussTedrake, 2020-11-01:
thanks jeremy.  your test case is good.  i can reproduce the failure using that test.
I do have a `$HOME/.config/matplotlib`, but it is empty.  `$HOME/.cache/matplotlib` looks similarly benign (just some font cache).  I tried setting the `MPLCONFIGDIR=` and `MPLCONFIGDIR=blah` in various places, but nothing resolved the issue.

RussTedrake, 2020-11-01:
I definitely have `~/.local` leaking into my path... even when I do
```
russt@Puget-179850-01:~$ export PYTHONNOUSERSITE=1
russt@Puget-179850-01:~$ export PYTHONPATH=""
russt@Puget-179850-01:~$ python3 -c "import sys; print(sys.path)"
['', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/usr/local/lib/python3.6/dist-packages', '/home/russt/.local/lib/python3.6/site-packages', '/usr/lib/python3/dist-packages']
```
it doesn't *appear* to be matplotlib's fault, really?

jwnimmer-tri, 2020-11-01:
Here's what I see -- PYTHONNOUSERSITE is respected:
```console
jwnimmer@call-cps:~$ mkdir -p .local/lib/python3.6/site-packages
jwnimmer@call-cps:~$ python3 -c "import sys; print(sys.path)"
['', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/home/jwnimmer/.local/lib/python3.6/site-packages', '/usr/local/lib/python3.6/dist-packages', '/usr/lib/python3/dist-packages']
jwnimmer@call-cps:~$ env PYTHONNOUSERSITE=1 python3 -c "import sys; print(sys.path)"
['', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/usr/local/lib/python3.6/dist-packages', '/usr/lib/python3/dist-packages']
```

Note this text from the `site` docs:
> A path configuration file is a file whose name has the form name.pth and exists in one of the four directories mentioned above; its contents are additional items (one per line) to be added to sys.path.

Can you check around your drive for pth files?

RussTedrake, 2020-11-01:
will look.  also i've learned that...
```
russt@Puget-179850-01:~$ python3 -s -c "import sys; print(sys.path)"
['', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/usr/local/lib/python3.6/dist-packages', '/home/russt/.local/lib/python3.6/site-packages', '/usr/lib/python3/dist-packages']
russt@Puget-179850-01:~$ python3 -S -c "import sys; print(sys.path)"
['', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload']
```

RussTedrake, 2020-11-01:
I have loads of `.pth` files in my bazel cache (from numerous projects), and then
```
./.local/lib/python3.6/site-packages/distutils-precedence.pth
./.local/lib/python3.6/site-packages/matplotlib-3.3.2-py3.6-nspkg.pth
```

jwnimmer-tri, 2020-11-01:
Anything in `/usr/...`?

RussTedrake, 2020-11-01:
```
russt@Puget-179850-01:~/.local/lib/python3.6/site-packages$ more distutils-precedence.pth 
import os; var = 'SETUPTOOLS_USE_DISTUTILS'; enabled = os.environ.get(var, 'stdlib') == 'local'; enabled and __import__('_distutils_hack').add_shim(); 
russt@Puget-179850-01:~/.local/lib/python3.6/site-packages$ more matplotlib-3.3.2-py3.6-nspkg.pth 
import sys, types, os;has_mfs = sys.version_info > (3, 5);p = os.path.join(sys._getframe(1).f_locals['sitedir'], *('mpl_toolkits',));importlib = has_mfs and __import__('importlib.util');has_mfs and __import__('importlib.machinery');m = has_mfs and sys.modules.setdefault('mpl_toolkits', importlib.util.module_from_spec(importli
b.machinery.PathFinder.find_spec('mpl_toolkits', [os.path.dirname(p)])));m = m or sys.modules.setdefault('mpl_toolkits', types.ModuleType('mpl_toolkits'));mp = (m or []) and m.__dict__.setdefault('__path__',[]);(p not in mp) and mp.append(p)
```
that `.pth` file might be related to the mpl_toolkits issue, but i *think* the root problem is happening before this.. .that my PYTHONNOUSERSITE is not being respected...

RussTedrake, 2020-11-01:
```
russt@Puget-179850-01:/usr$ find . -iname "*.pth"
./lib/wx/python/wx3.0.pth
./lib/wx/python/wx.pth
./lib/python2.7/dist-packages/pygtk.pth
./lib/python2.7/dist-packages/matplotlib-2.1.1-nspkg.pth
./lib/python2.7/dist-packages/zope.interface-4.3.2-nspkg.pth
./lib/python2.7/dist-packages/wx.pth
./lib/python3/dist-packages/matplotlib-2.1.1-nspkg.pth
./lib/python3/dist-packages/zope.interface-4.3.2-nspkg.pth
./lib/python3/dist-packages/ruamel.yaml-0.15.34-nspkg.pth
./local/lib/python2.7/dist-packages/configparser-3.5.0-py2.7-nspkg.pth
./local/lib/python2.7/dist-packages/matplotlib-2.0.2-py2.7-nspkg.pth
./local/lib/python3.6/dist-packages/easy-install.pth
./local/lib/python3.6/dist-packages/setuptools.pth
```

RussTedrake, 2020-11-01:
```
russt@Puget-179850-01:/usr$ more local/lib/python3.6/dist-packages/setuptools.pth 
/home/russt/.local/lib/python3.6/site-packages
```
that must be the culprit!

RussTedrake, 2020-11-01:
confirmed.
```
russt@Puget-179850-01:/usr/local/lib/python3.6/dist-packages$ sudo mv setuptools.pth setuptools.pth.bk
russt@Puget-179850-01:/usr/local/lib/python3.6/dist-packages$ python3 -s -c "import sys; print(sys.path)"
['', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/usr/local/lib/python3.6/dist-packages', '/usr/lib/python3/dist-packages']
russt@Puget-179850-01:/usr/local/lib/python3.6/dist-packages$ python3 -S -c "import sys; print(sys.path)"
['', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload']
```

RussTedrake, 2020-11-01:
now.. the question is... is simply leaving it removed going to screw up something else?

RussTedrake, 2020-11-01:
My original two use cases now pass (I believe)
```
russt@Puget-179850-01:~/drake$ bazel test //bindings/pydrake/systems:py/planar_scenegraph_visualizer_test --nocache_test_results --test_output=streamed
Starting local Bazel server and connecting to it...
INFO: Invocation ID: 2b80db5c-e2c6-4fd4-af20-2b104dedad59
WARNING: Streamed test output requested. All tests will be run locally, without sharding, one at a time
INFO: Analyzed target //bindings/pydrake/systems:py/planar_scenegraph_visualizer_test (175 packages loaded, 22139 targets configured).
INFO: Found 1 test target...
..[2020-11-01 15:37:47.192] [console] [warning] Currently MultibodyPlant does not handle joint limits for continuous models. However some joints do specify limits. Consider setting a non-zero time step in the MultibodyPlant constructor; this will put MultibodyPlant in discrete-time mode, which does support joint limits.
[2020-11-01 15:37:47.192] [console] [warning] Joints that specify limits are: `iiwa_joint_1`, `iiwa_joint_2`, `iiwa_joint_3`, `iiwa_joint_4`, `iiwa_joint_5`, `iiwa_joint_6`, `iiwa_joint_7`.
[2020-11-01 15:37:47.529] [console] [warning] Currently MultibodyPlant does not handle joint limits for continuous models. However some joints do specify limits. Consider setting a non-zero time step in the MultibodyPlant constructor; this will put MultibodyPlant in discrete-time mode, which does support joint limits.
[2020-11-01 15:37:47.529] [console] [warning] Joints that specify limits are: `iiwa_joint_1`, `iiwa_joint_2`, `iiwa_joint_3`, `iiwa_joint_4`, `iiwa_joint_5`, `iiwa_joint_6`, `iiwa_joint_7`.
...
----------------------------------------------------------------------
Ran 5 tests in 0.867s

OK
Target //bindings/pydrake/systems:py/planar_scenegraph_visualizer_test up-to-date:
  bazel-bin/bindings/pydrake/systems/py/planar_scenegraph_visualizer_test
INFO: Elapsed time: 10.518s, Critical Path: 3.48s
INFO: 2 processes: 2 linux-sandbox.
INFO: Build completed successfully, 2 total actions

INFO: Build completed successfully, 2 total actions

russt@Puget-179850-01:~/drake$ python3 -c "import sys; print(sys.path)"
['', '/home/russt/underactuated', '/home/russt/manipulation', '/home/russt/drake-install/lib/python3.6/site-packages', '/home/russt/drake', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/home/russt/.local/lib/python3.6/site-packages', '/usr/local/lib/python3.6/dist-packages', '/usr/lib/python3/dist-packages']
```
and I can run tests in both the manipulation and drake repos without any pip installing/uninstalling.  hoorah!  thanks @jwnimmer-tri .

jwnimmer-tri, 2020-11-01:
It's unusual to have meaningful python files within `/usr/local`.  A `pip3 install` as a _user_ account would be in `~/.local`.  Is there any workflow you use that requires `sudo pip3 install`?  I think that's where the problem came from.

RussTedrake, 2020-11-01:
i see.  i had that happening on accident in transition for my `manipulation` files (I had added the pip call to the installation script for CI, and run it locally to test).  i've purged that, but this machine did run it once.  is there a recommended way to disinfect?

RussTedrake, 2020-11-01:
i've purged them all manually with `sudo pip3 uninstall` and then `rm`ing the fragments left over in `/usr/local/lib/python3.6/dist-packages/`.  That directory is now empty for me (but it still exists).  Seem reasonable?

jwnimmer-tri, 2020-11-01:
TRI has internal advice for cleanup.  I've emailed you the link.

RussTedrake, 2020-11-19:
FWIW -- I think this issue is still real.  When I properly add matplotlib > 3.1.3 to bazel via `rules_python` and `pip_install`, it has the same problem.  Effectively `rules_python` is exposing only the `matplotlib` package to the source, but not the `mpl_toolkits` package (which is not, in and of itself, a `pip` package).  So python goes and finds the system version instead.  As of 3.2, that seems to break things.

I am currently ok pinning to 3.1.3, but maybe we should announce this issue upstream?

jamiesnape, 2020-11-19:
At least on my machine, I have `matplotlib-3.3.3-py3.9-nspkg.pth` that adds the appropriate `mpl_toolkits` package to the `PYTHONPATH`. Bazel almost certainly ignores that.

RussTedrake, 2020-11-19:
yes.  it has that file, too.  I agree that bazel ignoring that is the fundamental problem.

jwnimmer-tri, 2020-11-20:
I agree that this should be an issue, but it seems like it should be a https://github.com/RussTedrake/manipulation issue, not a Drake issue?  Drake does not use `pip_install` from `rules_python`.

RussTedrake, 2021-04-18:
I feel that this is an issue for drake because our bazel builds are not hermetic w/ respect to pip.  Any user can run `pip install matplotlib` and have a broken drake build.

jwnimmer-tri, 2021-04-18:
Can you provide a reproduction recipe for the failure you think is relevant?

The bulk of the investigation above related to a pth file hiding in `/usr/local/lib/python3.6/dist-packages`.  For that particular failure mechanism, there is probably not anything we can do to fix or avoid the problem.  The best we could do is have `install_prereqs` detect the situation and refuse to continue.

jamiesnape, 2021-04-19:
>  I feel that this is an issue for drake because our bazel builds are not hermetic w/ respect to pip

AFAIK No one's bazel builds are hermetic with respect to any external mechanism for installing Python dependencies. The blame (whether justified or not) goes 3-6 ways. bazel core and/or rules_python because they are not hermetic, matplotlib because it allows toolkits from a different version to be loaded, drake and manipulation/underactuated because they can both be broken. The order in which the problem could/should be fixed pretty much goes in the order I have listed the projects.

jamiesnape, 2021-04-19:
FTR I do not think matplotlib is doing a huge amount wrong. How drake, manipulation, and underactuated deal with the issue is a matter of preference and taste, I think.

RussTedrake, 2021-09-06:
I think we've agree this is a WON'T FIX.

closed 2021-09-06 by RussTedrake

RussTedrake, 2021-11-17:
FTR. This bit me again this week on my mac github actions CI.  homebrew just bumped it's matplotlib version from 3.4.3 to 3.5.0, which has breaking changes.  I needed to upgrade my pegged revision of matplotlib in my course repos to 3.5.0

RussTedrake, 2021-12-02:
Just to summarize the advice (since it just bit me *again* on focal):

Drake's `install_prereqs` installs matplotlib via `brew install matplotlib` on mac, and `apt install python3-matplotlib` on ubuntu.  If one is going to add matplotlib also to the requirements file processed by `rules_python`, then at least the *minor* version number in that requirements file must match the minor version number installed via the `brew` or `apt` method.