RobotLocomotion/drake #14258: geometry_py bindings for SceneGraphInspector variables names don't match c++ variable names

RussTedrake, 2020-10-28:
we have a lot of instances of e.g. bindings of the form
```
GetFrameId(geometry_id=)
```
when the c++ is 
```
GetFrameId(id=)
```
Might require some annoying deprecation gymnastics.  :-(  marked as low priority.

cc @EricCousineau-TRI , @mpetersen94

SeanCurtis-TRI, 2020-10-28:
[I noticed the same yesterday in a current PR](https://github.com/RobotLocomotion/drake/pull/14247#pullrequestreview-517887987) when I was updating bindings.

The solution I took was to simply change the name of the C++ to match the python. There's zero cost to do that in C++. It's not my *favorite* solution as I favor the terser `id` when there is only a single `id`. However, in python, it makes sense to load up on semantics.  So, I gave up the terseness in C++ for greater type clarity in python (which is why I presume they were bound that way). So, very easily solvable with little cost.

SeanCurtis-TRI, 2020-10-28:
The distinction in the parameter name has particular importance when you have two methods `GetName()` where one takes a `FrameId`, the other a `GeometryId`, and in C++, the parameters are both called `id`. That simply collapses in python bindings. So, I'll go ahead and make sure that the public API uses `frame_id` and `geometry_id` so that there are reasonable handles for python invocation.

RussTedrake, 2020-10-28:
I thought that might be true, but my PR #14259 uses the names "id", and seems to work just fine.

SeanCurtis-TRI, 2020-10-28:
I looked there. And I didn't see any instances where you had an overloaded method with different id types but same names. Was there a particular method overload you had in mind?

SeanCurtis-TRI, 2020-10-28:
Ah....nice to know.

That said, I still suspect that being more verbose should be preferred. Given the overload and the whole point of named parameters, it seems that it helps code readability if the parameter names are actually distinct. And it's much easier to change C++ parameter names to match the python names.

RussTedrake, 2020-10-28:
both `BelongsToSource` and `GetOwningSourceName` in my PR have that feature, too.  (it's like half the PR ;-)

SeanCurtis-TRI, 2020-10-28:
`BelongsToSource` is using parameter names `geometry_id` and `frame_id`. But `GetOwningSourceName` definitely uses `id` in both cases. My eyes must have skimmed over that one because it immediately followed `BelongsToSource`.

So, the key part of this conversation is, what do we want the solution to be? 

1. What should the final state look like?
   - I'd advocate using `geometry_id`, `source_id`, and `frame_id` as parameter names so they come through to the python API. At least for those APIs where the type of the id isn't clear from the name of the method (e.g., `FramesForSource` probably can be `id`, but `GetName` should be `frame_id` and `geometry_id`).
  - Alternatively, terse holds appeal. And as overloads are handled correctly, `id=geometry_id` or `id=my_frame_id` arguably handles the burden of communicating semantics. So, I'm open to arguments that we should push to simply `id` where there's a single identifier-type parameter.
2. What's the mechanism?
  - Where C++ changes, it's a free change.
  - Where C++ and python APIs currently have `id` (e.g., `GetOwningSourceName`), do we need to offer yet another overload and deprecate the old?

jwnimmer-tri, 2020-10-28:
When the user wants terse, they can just pass the argument as a positional value `GetOwningSourceName(some_thing)`.  Calls that choose to use named for the argument do so for clarity or robustness, which to me argues for the more specific names.  It might make the docs read more easily to use the longer names, depending on your point of view.

EricCousineau-TRI, 2020-10-28:
I agree with Jeremy's assessment, esp. given that using `geometry_id`, `frame_id` doesn't seem to incur deprecation overhead?

SeanCurtis-TRI, 2020-10-28:
I've got a plan:

## The end goal

For the public API of both `SceneGraphInspector` and `QueryObject`, parameters that are of `FooId` type will all be named `foo_id`. It'll be nicely uniform, clearly communicate state and, if it is too verbose, as @jwnimmer-tri pointed out, the python user can always just omit the keyword.

## The plan

As we survey the APIs and consider those that don't already match the desired schema, we divide them into three categories:

  1. Methods that have not yet been bound in master.
  2. Methods that have been bound in master, where the python bindings use the goal schema, but C++ does not.
  3. Methods that have been bound and don't match the schema.

tl;dr We already have a PR in flight (#14259). It is binding all of category 1. We simply have it resolve Category 1 at the same time. It can also easily resolve Category 2 (just a case of updating C++ parameter names and doxygen references). Category 3 will be done in a follow-up PR.

### Category 1: Methods that have not been bound in master yet.

  __Action__: Simply rename the parameters in C++ to match the schema (updating doxygen) and bound them as such.

  - SceneGraphInspector::NumFramesForSource(SourceId id)
  - SceneGraphInspector::FramesForSource(SourceId id)
  - SceneGraphInspector::GetFrameGroup(FrameId id)
  - SceneGraphInspector::NumGeometriesForFrame(FrameId id)
  - SceneGraphInspector::NumGeometriesForFrameWithRole(FrameId id, Role role)
  - SceneGraphInspector::GetFrameId(GeometryId id)
  - SceneGraphInspector::CollisionFiltered(GeometryId id1, GeometryId id2)
  - SceneGraphInspector::GetOwningSourceName(FrameId id)
  - SceneGraphInspector::GetOwningSourceName(GeometryId id)
  - SceneGraphInspector::GetName(FrameId id)
  - SceneGraphInspector::GetGeometryIdByName(FrameId id, Role role, const std::string& name)
  - SceneGraphInspector::GetFrameId(GeometryId id)

### Category 2: Methods that have previously been bound; python supports schema, C++ does not.

  __Action__: Simply rename the parameters in C++ to match the schema (updating doxygen).

  - SceneGraphInspector::GetName(GeometryId id)
  - SceneGraphInspector::GetShape(GeometryId id)
  - SceneGraphInspector::GetPoseInParent(GeometryId id)
  - SceneGraphInspector::GetPoseInFrame(GeometryId id)
  - SceneGraphInspector::GetProximityProperties(GeometryId id)
  - SceneGraphInspector::GetIllustrationProperties(GeometryId id)
  - SceneGraphInspector::GetPerceptionProperties(GeometryId id)
  - SceneGraphInspector::CloneGeometryInstance(GeometryId id)

### Category 3: Methods that have been bound and don't match the schema

The methods in this category have been partitioned by the action to be taken:

#### Group 1: Methods whose python parameter name needs to change, but we really want to change the method name, anyways.

  __Action__: Deprecate old function in favor of new function with new name. This requires deprecating the C++ as well.

  - QueryObject::X_WF(FrameId id)  --> QueryObject::PoseInWorld(FrameId frame_id)
  - QueryObject::X_PF(FrameId id)  -->  QueryObject::PoseInParent(FrameId frame_id)
  - QueryObject::X_WG(FrameId id)  --> QueryObject::PoseInWorld(GeometryId geometry_id)
  - SceneGraphInspector::GetSourceName(SourceId id) --> SceneGraphInspector::GetName(SourceId source_id) (this matches the other GetName() variants).

#### Group 2: Methods whose python parmeter name needs to change, but we want to *keep* the name.

__Action__: For Class::Method(FooId id):
    1. Rename C++ parameter name to `foo_id`.
    2. Inject *new* binding *before* the old binding: The difference is `py::arg("id")` --> `py::arg("foo_id")`.
    3. Deprecate the old binding.
    4. Augment tests to show that new binding is good, old binding is deprecated and the invocation without keywords goes to the new.

  - SceneGraphInspector::SourceIsRegistered(SourceId id)
  - QueryObject::ComputeSignedDistancePairClosestPoints(GeometryId id_A, GeometryId id_B)

PR #14280 (2020-11-03): Clean up and deprecation of SceneGraphInspector/QueryObject bindings
As per issue 14258, we want all public APIs that take Identifiers as parameters to have names representing their id type (source_id and not simply id). This handles what is termed "Category 3" APIs in that issue. This includes deprecations of some APIs that have been renamed and some deprecations on only the bindings.

resolves #14258

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14280)
closed 2020-11-03 by rpoyner-tri