RobotLocomotion/drake #14268: Create missing method Multibody::CalcCenterOfMassTranslationalVelocity()

mitiguy, 2020-10-30:
Vector3\<T\> CalcCenterOfMassTranslationalVelocityInWorld(const systems::Context\<T\>& context) const;  and/or
Vector3\<T\> CalcCenterOfMassTranslationalVelocity(const systems::Context\<T\>& context,  
                                                   const Frame\<T\>& frame_A, const Frame\<T\>& frame_E) const;
---------------------------------------------------------------------------------------------
Note: Two related MultibodyPlant methods are:
Vector3\<T\> CalcCenterOfMassPosition(const systems::Context\<T\>& context) const;
Vector3\<T\> CalcCenterOfMassPosition(const systems::Context\<T\>& context,
      const std::vector<ModelInstanceIndex>& model_instances) const;
---------------------------------------------------------------------------------------------
Note:  PR# #12839 added a similar multibodyPlant method with signature:
void CalcJacobianCenterOfMassTranslationalVelocity( const systems::Context\<T\>& context, 
      JacobianWrtVariable with_respect_to,  const Frame\<T\>& frame_A, const Frame\<T\>& frame_E,
      EigenPtr\<Matrix3X\<T\>\> Js_v_ACcm_E) const;

mitiguy, 2020-11-06:
PR #14297 should resolve issue #14268.

mitiguy, 2020-11-12:
@amcastro-tri, @sherm1 @joemasterjohn 
It would be helpful to get your input on the API design for CalcCenterOfMassTranslationalVelocity...
The choice for this API will also affect the API for upcoming method CalcCenterOfMassTranslationalAcceleration...

Some possible options for the API:

Vector3\<T\> CalcCenterOfMassTranslationalVelocityInWorld(const systems::Context\<T\>& context) const; 

Vector3\<T\> CalcCenterOfMassTranslationalVelocity(const systems::Context\<T\>& context,  
                                                   const Frame\<T\>& frame_A, const Frame\<T\>& frame_E) const;

OR YOUR SUGGESTION

Reminder of related APIs that are currently in Drake (which can be changed if needed).
Vector3<T> CalcCenterOfMassPosition(const systems::Context<T>& context) const;

void CalcJacobianCenterOfMassTranslationalVelocity( const systems::Context<T>& context,
JacobianWrtVariable with_respect_to, const Frame<T>& frame_A, const Frame<T>& frame_E,
EigenPtr<Matrix3X<T>> Js_v_ACcm_E) const;

sherm1, 2020-11-12:
My favorite API set:
```c++
Vector3 CalcCenterOfMassPositionInWorld(const systems::Context& context) const;
Vector3 CalcCenterOfMassVelocityInWorld(const systems::Context& context) const;
Vector3 CalcCenterOfMassAccelerationInWorld(const systems::Context& context) const;
```
I don't think we need to provide results in any other frame since (a) hard to imagine why anyone would want that, and (b) if they do we can just point them to the existing shift and re-express methods. Also, the "InWorld" suffix leaves open the possibility of adding more-general methods if needed.

It might also be necessary for performance reasons to provide combined methods like CalcCenterOfMassPositionAndVelocity() since those can be made much faster than calls to separate methods. I would not advocate doing that now, though!

amcastro-tri, 2020-11-16:
I like @sherm1's proposal. It is very true that most likely no one will want those quantities in any other frame than the world frame. 
The world signatures look concise and clean.

joemasterjohn, 2020-11-16:
I defer to @sherm1 and @amcastro-tri on the API design. Since the current PR (#14297) is already using:
```
Vector3<T> CalcCenterOfMassTranslationalVelocity(const systems::Context<T>& context,
const Frame<T>& frame_A, const Frame<T>& frame_E) const;
```
an option would be to make that method private (with the option to make it a public API later) and have the public method use it.
Not sure of the usefulness at this point.

joemasterjohn, 2020-11-16:
Actually sorry my bad, you don't have to make any private member functions on `MultibodyPlant`. Since the plant method is just calling the underlying implementation in `MultibodyTree`, you can just change the public plant API and leave the tree API as is.

sherm1, 2020-11-16:
Although ... we would then have unused (and potentially untested) extra complexity in MBTree which would then lead either to similar unused complexity for CoMTranslationalAcceleration or an incompatibility between Velocity and Acceleration MBTree APIs. I don't think "sunk cost" is a good reason to keep the extra code around if we don't have any other reason to do so.

mitiguy, 2020-11-17:
I think the conversation has converged to the following public API.
Let me know if this is incorrect.

Vector3\<T\> CalcCenterOfMassPositionInWorld(const systems::Context& context) const;
Vector3 \<T\> CalcCenterOfMassVelocityInWorld(const systems::Context& context) const;
Vector3\<T\>  CalcCenterOfMassAccelerationInWorld(const systems::Context& context) const;

Vector3\<T\> CalcCenterOfMassPositionInWorld(const systems::Context<T>& context,
const std::vector& model_instances) const;
Vector3\<T\> CalcCenterOfMassVelocityInWorld(const systems::Context<T>& context,
const std::vector& model_instances) const;
Vector3\<T\> CalcCenterOfMassAccelerationInWorld(const systems::Context<T>& context,
const std::vector& model_instances) const;

Note: I assume I/we will eventually deprecate the following:
Vector3\<T\> CalcCenterOfMassPosition(const systems::Context& context) const;
Vector3\<T\> CalcCenterOfMassPosition(const systems::Context<T>& context,
const std::vector& model_instances) const;

sherm1, 2020-11-17:
Looks great. I would suggest deprecating CalcCenterOfMassPosition() in the same PR in favor of CalcCenterOfMassPositionInWorld() to avoid ever having two names for the same thing.

mitiguy, 2021-01-12:
Resolved as PR# 14354 merged with methods and test for MultibodyPlant::CalcCenterOfMassTranslationalVelocityInWorld().
Will open a new issue to deal with Sherm's suggestion to deprecate CalcCenterOfMassPosition() in favor of CalcCenterOfMassPositionInWorld() (avoid having two names for the same thing).

mitiguy, 2021-01-16:
Specifically, created issue #14537 and PR# 14538 to deprecate CalcCenterOfMassPosition() in favor of CalcCenterOfMassPositionInWorld().

closed 2021-01-16 by mitiguy