RobotLocomotion/drake #14286: Testing MOSEK intermittently has license failures when using MOSEK floating license

mfry90, 2020-11-04:
It appears that running Bazel MOSEK tests provide spotty results when using a MOSEK floating license. I'm guessing because the MOSEK floating license uses a license server to checkout a license when the MOSEK solver is called. This behaviour won't be seen when relying on personal academic or node-locked MOSEK licenses, but it's happening on my commercial floating license.

To reproduce, replace mosek.lic with a license obtained from a floating license or institutional academic license (see [https://docs.mosek.com/9.2/licensing/quickstart.html](url)).
```
bazel test --config mosek --test_tag_filters=mosek --cache_test_results=no //...
```
Returns:
```
==================== Test output for //solvers:non_convex_optimization_util_test (shard 1 of 4):
Using drake_cc_googletest_main.cc

Note: This is test shard 1 of 4.
[==========] Running 6 tests from 3 test suites.
[----------] Global test environment set-up.
[----------] 1 test from DecomposeNonConvexQuadraticForm
[ RUN      ] DecomposeNonConvexQuadraticForm.Test0
[       OK ] DecomposeNonConvexQuadraticForm.Test0 (0 ms)
[----------] 1 test from DecomposeNonConvexQuadraticForm (0 ms total)

[----------] 4 tests from TestRelaxNonConvexQuadraticConstraintInTrustRegion
[ RUN      ] TestRelaxNonConvexQuadraticConstraintInTrustRegion.Test0
[       OK ] TestRelaxNonConvexQuadraticConstraintInTrustRegion.Test0 (0 ms)
[ RUN      ] TestRelaxNonConvexQuadraticConstraintInTrustRegion.SolveProblem1
unknown file: Failure
C++ exception with description "Could not acquire MOSEK license." thrown in the test body.
[  FAILED  ] TestRelaxNonConvexQuadraticConstraintInTrustRegion.SolveProblem1 (3 ms)
[ RUN      ] TestRelaxNonConvexQuadraticConstraintInTrustRegion.ZeroQ1Test0
unknown file: Failure
C++ exception with description "Could not acquire MOSEK license." thrown in the test body.
[  FAILED  ] TestRelaxNonConvexQuadraticConstraintInTrustRegion.ZeroQ1Test0 (3 ms)
[ RUN      ] TestRelaxNonConvexQuadraticConstraintInTrustRegion.ZeroQ2Test0
[       OK ] TestRelaxNonConvexQuadraticConstraintInTrustRegion.ZeroQ2Test0 (318 ms)
[----------] 4 tests from TestRelaxNonConvexQuadraticConstraintInTrustRegion (324 ms total)

[----------] 1 test from TestRelaxNonConvexQuadraticConstraintInTrustRegionInfeasible
[ RUN      ] TestRelaxNonConvexQuadraticConstraintInTrustRegionInfeasible.SolveInfeasibleProblem1
unknown file: Failure
C++ exception with description "Could not acquire MOSEK license." thrown in the test body.
...
```
I've noted that MOSEK will run successfully during some tests, and fail with `Could not acquire MOSEK license` on others, I'm guessing because the floating license is checked out for a different test. When I replace my mosek.lic with a trial license (similar to a personal academic one), I pass all the tests.

I don't think this is a big deal, not even sure if it needs a fix, but an FYI for other MOSEK floating license users trying to test Drake's MOSEK calls.

jwnimmer-tri, 2020-11-04:
If there are fewer license seats available than full test concurrency would require, then you can set `--local_test_jobs=1` (or 2, or ...) to limit how many tests are run concurrently.  The max `--jobs` you've configured will still be used during compilation, but fewer _tests_ will be run concurrently.

https://docs.bazel.build/versions/master/command-line-reference.html

mfry90, 2020-11-04:
Great, now all the tests are passing, thanks!

closed 2020-11-04 by jwnimmer-tri