RobotLocomotion/drake #14290: panda_link8 is detached from arm?

EricCousineau-TRI, 2020-11-04:
On drake @ 96e0cc0b94926f0725ba69f544fda6e9200b216c (was from Anzu):

```py
from pydrake.all import *

builder = DiagramBuilder()
plant, scene_graph = AddMultibodyPlantSceneGraph(builder, time_step=0.005)
ConnectDrakeVisualizer(
    builder, scene_graph, role=Role.kProximity)
arm = Parser(plant).AddModelFromFile(
    FindResourceOrThrow("drake/manipulation/models/franka_description/urdf/panda_arm.urdf")
)
plant.WeldFrames(
    plant.world_frame(),
    plant.GetFrameByName("panda_link0"),
    RigidTransform(p=[0, 0, 0.5]),
)
plant.Finalize()

diagram = builder.Build()
Simulator(diagram).Initialize()
```

Output:
> [image]

@sammy-tri or @calderpg-tri - y'all seen this?

EricCousineau-TRI, 2020-11-04:
Er, it looks like it should be fine???
https://github.com/RobotLocomotion/drake/blob/96e0cc0b94926f0725ba69f544fda6e9200b216c/manipulation/models/franka_description/urdf/panda_arm.urdf#L509-L521

EricCousineau-TRI, 2020-11-04:
While on the phone w/ @SeanCurtis-TRI, we see that `DrakeVisualizer` has different output:
```
drake_viz = builder.AddSystem(DrakeVisualizer(params=DrakeVisualizerParams(role=Role.kProximity)))
builder.Connect(
    scene_graph.get_query_output_port(),
    drake_viz.query_object_input_port(),
)
```
> [image]

EricCousineau-TRI, 2020-11-04:
Per convo, this seems like it's not a huge problem - only the soon-to-be-deprecated `ConnectDrakeVisualizer` encounters this, so whatevs. That function / setup is deprecated (and we all use `QueryObject`-based visualization), this issue can be closed.

SeanCurtis-TRI, 2020-11-04:
I don't see why this is still open. This is a combination of "won't fix" (re: `ConnectDrakeVisualizer`) and replace with code that is already available that *does* work. The only thing missing is the sugar that is currently in flight (#14282): `DrakeVisualizer::AddToBuilder(&builder, &lcm)`. What's the point of keeping this open?

EricCousineau-TRI, 2020-11-04:
To let people know it's something to not be worried about if they stumble across it? But yeah, I'm fine w/ closing it for now.

closed 2020-11-04 by EricCousineau-TRI