RobotLocomotion/drake #14291: Is it possible to organize/compact documentation for setter/getter methods for parameters?

mitiguy, 2020-11-04:
During review of reflected inertia PR# [14167](https://reviewable.io/reviews/RobotLocomotion/drake/14167).  This issue was initially discussed with @joemasterjohn  and @SeanCurtis-TRI who encouraged making a Github issue.

The example code below shows gear ratio methods (2 setter and 2 getter methods).  In this PR, there is another setter/getter method for rotor inertia.  Hence for the two parameters in this PR, there are a total of 8 setter/getter methods.
Reminder: More parameters have been recently added, with more soon -- each with additional setter/getter methods.  Hence, a plan for organized, efficient, concise documentation that helps both developers and end-users would be helpful.

Focusing attention on only the gear ratio methods:

  /// Gets the default value for this actuator's gear ratio.
  /// See @ref reflected_inertia.
  double default_gear_ratio() const { return default_gear_ratio_; }

  /// Sets the default value for this actuator's gear ratio.
  /// See @ref reflected_inertia.
  void set_default_gear_ratio(double gear_ratio);

  /// Returns the associated gear ratio value for this actuator, stored in
  /// `context`.
  /// See @ref reflected_inertia.
  const T& gear_ratio(const systems::Context<T>& context) const;

  /// Sets the associated gear ratio value for this actuator in `context`.
  /// See @ref reflected_inertia.
  void SetGearRatio(systems::Context<T>* context, T gear_ratio);

In addition to the documentation here, there is a need for a centralized discussion of how parameters work, e.g., the interaction between default parameter vs. those set through context.  So, the documentation for these methods may eventually include a link to that centralized discussion of how parameters work.

sherm1, 2020-11-06:
We should standardize the naming conventions for methods dealing with default & in-context parameters. I like the names used above and propose that we follow that conventions for all parameters. An audit to bring all the names into compliance should be part of this project.

joemasterjohn, 2020-11-06:
I'll take inventory. So far the classes that have been parameterized are:
- `RigidBody`
- `FixedOffsetFrame`
- `LinearBushingRollPitchYaw`
- `JointActuator`

`MultibodyPlant` already has a section titled [Parameters](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#mbp_parameters). I think it would be ideal to make that the centralized place for the discussion on how parameters work. Then each class that makes use of the MBElement parameters API can reference it.

jwnimmer-tri, 2020-11-09:
I'd also like to advocate for scoping the SetRandomContext affordance into this consolidation as well -- at least for the design, even if we don't implement it right away.

See, for example,
```c++
  void PlanarJoint<T>::set_random_pose_distribution(
      const Vector2<symbolic::Expression>& p_FoMo_F,
      const symbolic::Expression& theta) {
```

We should plan for things like the `mass` parameters to similarly allow for Expression to populate a random value.

sherm1, 2020-11-09:
Do you mean something like this?
```c++
  // Default distribution gets built into the System.
  void PlanarJoint<T>::set_default_random_pose_distribution(
      const Vector2<symbolic::Expression>& p_FoMo_F,
      const symbolic::Expression& theta);

  // Override default distribution in this Context.
  void PlanarJoint<T>::SetRandomPoseDistribution(
      systems::Context<T>* context,
      const Vector2<symbolic::Expression>& p_FoMo_F,
      const symbolic::Expression& theta) const;
```

jwnimmer-tri, 2020-11-09:
I'm not trying to specify a solution (the snippet I posed was an existing method).  I'm asking that any design proposal account for both the default _and random_ cases.

sherm1, 2020-11-09:
I haven't thought much about the random feature. I'm unclear whether we should logically have default/context _numerical_ and default/context _random_, i.e. 4 cases or default/random/context, i.e. 3 cases. (Your last comment sounds like the latter?)

jwnimmer-tri, 2020-11-09:
To sample the random distribution into a Context, I think we should have the user call `System::SetRandomContext` on the whole diagram.  We don't need a public API for the user to just sample one element at a time.  The per-element API only needs to allow for configuring the distribution, not sampling it.

sherm1, 2020-11-09:
> The per-element API only needs to allow for configuring the distribution, not sampling it.

How does that relate to the in-system (default) vs in-context naming convention here? Do we still need a default distribution and an in-context distribution?

jwnimmer-tri, 2020-11-09:
We definitely need the in-system storage (and therefore a method naming choice) for a random `Expression`.

I don't even know what it would mean to try to store a symbolic distribution for a parameter into a `Context<double>`.  I don't think you need to design for that.