RobotLocomotion/drake #14315: Capability to compute the Coriolis matrix.

amcastro-tri, 2020-11-10:
We can write the Coriolis bias term as `c(q, v) = C(q,v)⋅v`. MultibdodyPlant offers `MBP::CalcBiasTerm()` to compute the (lower case) `c(q,v)`. 
We want the capability to compute the (upper case) matrix `C(q,v)`.

Brian Acosta provided [this nice reference](https://arxiv.org/pdf/2010.01033.pdf) on a recursive algorithm to compute this matrix.
cc'in @mposa.

### Observations
 - Even though  `c(q,v)` is unique, the Coriolis matrix C(q,v) is not.
 - `c(q,v)` is quadratic in v and can be defined in terms of Christoffel symbols function of q only.
 - That is, each component of `c(q,v)` is a quadratic in v for which the coefficient matrix is symmetric.

This last observation give us a method to compute C(q,v) through differentiation (or so I believe, please double check my math). Using indicial notation we can see that `Cᵢⱼ = 1/2⋅∂cᵢ/∂vⱼ`.

sherm1, 2020-11-10:
cc @Brian-Acosta

sherm1, 2020-11-10:
cc @mitiguy

mitiguy, 2020-11-10:
From the documentation for MultibodyPlant::CalcBiasTerm( ... ), the equations are written as:
  M(q) v̇ + C(q, v) v = tau_app + ∑ (Jv_V_WBᵀ(q) ⋅ Fapp_Bo_W)

I may be missing something that was said earlier.
What is the underlying motivation for the need (or importance) in calculating capital C(q,v) ?

Brian-Acosta, 2020-11-11:
Thanks for opening this! This came up because I'm implementing a passivity based controller (http://www.roboticsproceedings.org/rss16/p077.pdf) and C(q, v) is needed to find the desired task space acceleration.

sherm1, 2022-05-31:
Thanks, Patrick! cc @mitiguy

mitiguy, 2025-03-17:
Hi @pwensing.    In the months ahead, I'll be working on Drake code that will eventually lead to computing the Coriolis matrix.  You had mentioned extra notes that might help direct this work.  If/when, you have a chance, it would be great to get them to guide my software development (perhaps by summer)? 

In our small world, our  Drake development team just finished a Google conference video, with @vincekurtz (Vincent Kurtz) reporting on his development progress.  He is a great asset!  Thank you for helping to train him.

Also, my son JohnPaul just returned Saturday to ND.  I sent along some fun cards (with a light-hearted  **F**=m**a** sticker) to you, Michael Seelinger, and Margaret Coad.  Hopefully, he completes the delivery!
FYI: My emails are paul.mitiguy@tri.global  and  mitiguy@stanford.edu.