RobotLocomotion/drake #14322: Yaml serialization of optional distribution vector fails (emits invalid yaml)

ggould-tri, 2020-11-11:
`YamlWriteArchive` serialization of `std::optional<DistributionVectorVariantX>`'s `DeterministicVector` arm emits invalid yaml consisting only of the type tag and without the value.

See the following additional test case:
https://github.com/RobotLocomotion/drake/compare/master...ggould-tri:demonstrate_stochastic_optional_fail#diff-b2d63c15aa78ba9735ccd7bc49429dac220c82a4bf34361c1f717fdfa5815074R432-R446

That test emits the failure:
```
common/schema/test/stochastic_test.cc:445: Failure
Expected equality of these values:
  writer.EmitString()
    Which is: "root:\n  optional_dist_vec: !DeterministicVector\n"
  R"""(root:
  optional_dist_vec: !DeterministicVector { value: [ 1.0, 2.0 ] }
)"""
    Which is: "root:\n  optional_dist_vec: !DeterministicVector { value: [ 1.0, 2.0 ] }\n"
With diff:
@@ -1,2 +1,2 @@
 root:
-  optional_dist_vec: !DeterministicVector\n
+  optional_dist_vec: !DeterministicVector { value: [ 1.0, 2.0 ] }\n

[  FAILED  ] StochasticTest.OptionalTest (2 ms)
```
As you can see, the emitted yaml for the variant element is just the string `!DeterministicVector` with no further content.

FYI @jwnimmer-tri

ggould-tri, 2020-11-11:
From local experimentation, I am finding that if I add another element after the variant it is not serialized at all, leading me to suspect that the writer is taking an early return between emitting the tag and its content rather than simply mis-serializing this structure.

ggould-tri, 2020-11-11:
Jeremy:  Assigning this to you as the SME; feel free to assign it back to me if you get overloaded.  I'm not blocked on this, although the behaviour is somewhat worrisome.

jwnimmer-tri, 2020-11-12:
The bug relates to having `DeterministicVector<-1>` as the tag.  When reading YAML, we discard the template argument while comparing types.  While writing, we fail to strip the template argument from the tag.  And, for some reason, writing a `<` in a tag name makes yaml lose its mind.  I'll work up a patch.

PR #14327 (2020-11-17): yaml: Correctly emit variants with templated inner types
We need to discard the template arguments; they are not valid tag syntax.

Also add a stochastic acceptance test that would have caught this issue.

Closes #14322.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14327)
closed 2020-11-17 by jwnimmer-tri