RobotLocomotion/drake #14325: [pydrake] Deprecating nested classes infects parent class documentation

SeanCurtis-TRI, 2020-11-12:
When a parent class (e.g., [`RgbdSensor`](https://github.com/RobotLocomotion/drake/blob/f2cdfd2f9855c7f6cfc6ab880c8e7d09df0c0cd4/systems/sensors/rgbd_sensor.h#L96) has a public nested class (e.g., ['CameraPoses'](https://github.com/RobotLocomotion/drake/blob/f2cdfd2f9855c7f6cfc6ab880c8e7d09df0c0cd4/systems/sensors/rgbd_sensor.h#L101), the act of deprecating the nested class bleeds into the parent class.

For example, given the [current code]()
```c++
class RgbdSensor final : public LeafSystem<double> {
 public:
  DRAKE_NO_COPY_NO_MOVE_NO_ASSIGN(RgbdSensor)

  /** Specifies poses of cameras with respect ot the sensor base `B`.  */
  struct CameraPoses {
    /** Pose of color camera `C` with respect to sensor base `B`. Defaults to
     the identity matrix.  */
    math::RigidTransformd X_BC;
    ...
```

The contents of `documentation_pybind.h` looks something like this:

```c++
Note:
    These depth sensor measurements differ from those of range data
    used by laser range finders (like DepthSensor), where the depth
    value represents the distance from the sensor origin to the
    object's surface.)""";
          // Symbol: drake::systems::sensors::RgbdSensor::CameraPoses
          struct /* CameraPoses */ {
            // Source: drake/systems/sensors/rgbd_sensor.h:101
            const char* doc =
R"""(Specifies poses of cameras with respect ot the sensor base ``B``.)""";
            // Symbol: drake::systems::sensors::RgbdSensor::CameraPoses::X_BC

```

However if we introduce deprecation on the nested class (a la):

```c++
class RgbdSensor final : public LeafSystem<double> {
 public:
  DRAKE_NO_COPY_NO_MOVE_NO_ASSIGN(RgbdSensor)

  /** Specifies poses of cameras with respect ot the sensor base `B`.  */
  struct DRAKE_DEPRECATED("2021-03-01",
                   "The constructors that take poses explicitly have been "
                   "deprecated. Pose is now part of the RenderCamera "
                   "interface. See the RenderCamera-based RgbdSensor "
                   "constructors.")
  CameraPoses {
    /** Pose of color camera `C` with respect to sensor base `B`. Defaults to
     the identity matrix.  */
    math::RigidTransformd X_BC;
```

We get the following pybind
```c++
...
Note:
    These depth sensor measurements differ from those of range data
    used by laser range finders (like DepthSensor), where the depth
    value represents the distance from the sensor origin to the
    object's surface.

Deprecated:
    The constructors that take poses explicitly have been deprecated.
    Pose is now part of the RenderCamera interface. See the
    RenderCamera-based RgbdSensor constructors. This will be removed
    from Drake on or after 2021-03-01.)""";
          // Symbol: drake::systems::sensors::RgbdSensor::CameraPoses
          struct /* CameraPoses */ {
            // Source: drake/systems/sensors/rgbd_sensor.h:106
            const char* doc_deprecated =
R"""(Specifies poses of cameras with respect ot the sensor base ``B``. */
(Deprecated.)

Deprecated:
    The constructors that take poses explicitly have been deprecated.
    Pose is now part of the RenderCamera interface. See the
    RenderCamera-based RgbdSensor constructors. This will be removed
    from Drake on or after 2021-03-01.)""";
...
```

The deprecated clause appears in both the `CameraPoses` documentation as well as the `RgbdSensor`. Furthermore, the struct member for `RgbdSensor` has been renamed from `RgbdSensor.doc` to `RgbdSensor.doc_deprecated`.

jwnimmer-tri, 2020-11-12:
Presumably, my macro-matching heuristics have failed somehow, near this point:

https://github.com/RobotLocomotion/drake/blob/f2cdfd2f9855c7f6cfc6ab880c8e7d09df0c0cd4/tools/workspace/pybind11/mkdoc.py#L220-L221

SeanCurtis-TRI, 2020-11-12:
The alluded to 5 lines seem to suggest the particular pitfall. Do those lines include whitespace? I might try injecting some useless whitespace just to find out. Or moving the definition down deeper into the parent class declarations.

jwnimmer-tri, 2020-11-12:
The `.line` in the code there is literally the source file line number, so yes -- I think whitespace would work around it.

EricCousineau-TRI, 2020-11-12:
FWIW You may want to look at these two files and play with them in a more constrained context:
https://github.com/RobotLocomotion/drake/blob/f2cdfd2f9855c7f6cfc6ab880c8e7d09df0c0cd4/tools/workspace/pybind11/test/sample_header.h
https://github.com/RobotLocomotion/drake/blob/f2cdfd2f9855c7f6cfc6ab880c8e7d09df0c0cd4/tools/workspace/pybind11/test/sample_header_documentation.expected.h

SeanCurtis-TRI, 2020-11-12:
Yep. One line of whitespace makes everything happy. But to keep the linter happy, spreading the doxygen comment over *two* lines is sufficient. So, my immediate pain is good, the general issue still stands.

jwnimmer-tri, 2022-09-29:
See also #17999.

jwnimmer-tri, 2024-03-17:
The mkdoc tooling is difficult enough that working around temporary broken-ness will not rise to the level of action anytime soon. Closing as "not planned".

closed 2024-03-17 by jwnimmer-tri