RobotLocomotion/drake #14326: Disable RenderEngineOspray on macOS

jwnimmer-tri, 2020-11-12:
Per discussion in #13253, it's going to get increasingly more difficult to maintain Ospray on macOS -- especially with the switch to ARM hardware.  We will make it Ubuntu-only to reduce our build-support engineering cost.

\CC @SeanCurtis-TRI

SeanCurtis-TRI, 2020-11-12:
We might consider yanking it completely. It's not used. It produced a single shiny video. But it doesn't have full implementaiton of `RenderEngine` and doesn't have any mechanism for fully specifying meaningful materials. Maybe retiring it right out and resurrecting it if/when it has concrete value again. The purpose it was intended to serve has largely been supplanted with external techniques.

jwnimmer-tri, 2020-11-12:
I'll defer to you on that front.  If you want to pull it, I have no objection.

SeanCurtis-TRI, 2020-11-12:
Can we defer the question to the next drake developers meeting? Or should we try a slack poll?

jwnimmer-tri, 2020-11-12:
If we can purge Ospray prior to the VTK 9 upgrade, that would save us some work.  So, the earlier we can get an answer and stark the deprecation clock, the better.

SeanCurtis-TRI, 2020-11-12:
What is that time frame?

jwnimmer-tri, 2020-11-12:
It depends on how fast other work happens.  But maybe in a month or less.

jwnimmer-tri, 2020-11-12:
But the bottom line is really that if you think Ospray is not something we're going to maintain, we should just drop it without even a survey.  The code is there for anyone to fork and keep alive on their own dime.

SeanCurtis-TRI, 2020-11-12:
Then, in that case, I'll do a quick slack poll and make sure I'm not misreading the currents.  If things are as I believe, I'll submit a PR at the beginning of next week to jettison it.

SeanCurtis-TRI, 2020-11-12:
(FTR Ultimately, I'd like to have that functionality -- an ability to create high fidelity images from our simulation -- but I think we're exploring enough other options that we don't have to persist the current partial effort in the name of that goal. If nothing else pans out, then we bring it back and flesh it out.)

SeanCurtis-TRI, 2020-11-12:
The poll question has been [posted](https://drakedevelopers.slack.com/archives/C0JNB2E3S/p1605214671040200) (and cross posted to sensors and visualization). I'll revisit it at the beginning of next week and take action accordingly.

jamiesnape, 2020-11-13:
VTK 9 will get you an NVIDIA OptiX renderer for free with the OSPRay one (still no good for Mac, though), but pulling it and resurrecting it may not be such a bad idea if it is incomplete and not really being used.

PR #14339 (2020-11-17): [render] Remove RenderEngineOspray entirely
Other than killing the class, its factory, bindings, and their tests, this also touches a fair amount of other sites in the code:

1. Documentation of various sorts have references to the render engine removed.
2. Build infrastructure removes explicit dependencies on OSPRay (and Embree).
3. Removed OSPRay-specific functionality from docker configuration.

resolves #14326

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14339)
closed 2020-11-17 by rpoyner-tri