RobotLocomotion/drake #14328: Withdrawing macOS Mojave (10.14) support

jwnimmer-tri, 2020-11-12:
We are planning on dropping support for macOS Mojave (10.14) support in future releases.

Our usual policy is to keep N, N-1 versions alive for macOS -- so the most recent Apple release, and the one immediately prior.

With Big Sur on the way (#13648) it's time to drop Mojave to make room for it.

jamiesnape, 2020-11-17:
No complaints so far here and I just checked that all our system dependencies are happy with Big Sur (`bazel` and `scipy` protested somewhat, but are good now), so I would propose we start disabling the CI jobs for Mojave this afternoon.

jwnimmer-tri, 2020-11-17:
SGTM.  We should also do a doc/developers.rst update to go along with that.

PR #14352 (2020-11-17): Remove support for macOS Mojave 10.14
Closes #14328.

----
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14352)
closed 2020-11-17 by rpoyner-tri