RobotLocomotion/drake #14330: Heap allocations and virtual function calls in inner loops of FemModel.

xuchenhan-tri, 2020-11-13:
Currently FemModel owns a std::vector of pointers to FemElement, and the global quantities (residual, tangent matrix, etc.) are assembled by building the corresponding quantities in FemElement through these pointers. Since the assembly is an inner loop operation, it may make sense to cut down heap allocations and virtual function calls as much as possible. One way to remove the virtual layer, as @amcastro-tri suggested, is to have FemModel own a std::vector `FemGroup<FemElementType>`, where each "group" would typically correspond to a model or a mesh. FemGroup will perform all the computations needed from its set of elements. This will optimize the layout of its elements and process them in bulk with code that knows the size of everything at compile time. Relates to #14302 .

amcastro-tri, 2020-11-16:
Something that occurred to me last week, is that we can as a first step, limit `FemModel` to only have a single type of element. For several reasons:
1. We'll use the FEM code as a library. An external layer (`MultibodyPlant` or some other system) will use it to build the necessary matrices to assemble the contact problem. If we happen to have, say, two models like a gripper and a pancake with different element types, those could be two different `FemModel`s managed by the application layer.
2. In this picture, `FemModel` would be, the possibly later to be named, `FemGroup`. In that plausible future we'd rename `FemModel` to `FemGroup` and then write a thin `FemModel` layer that owns the groups. But, do we need this now?

xuchenhan-tri, 2020-11-16:
@amcastro-tri, am I correct in understanding that this the same general idea as the one described in the OP of this issue, but with the order of implementing FemModel and FemGroup reversed? In F2F, we said we would continue to flesh out the components of FemModel as structured now, and revisit this issue when we have a more complete picture (e.g. when linear elasticity is functional).

amcastro-tri, 2020-11-17:
All that is correct @xuchenhan-tri . Don't worry, I am not pumping the priority up, but only dumping my thoughts while fresh, which is part of the reason we have issues for.

xuchenhan-tri, 2022-04-18:
The new FEM design has landed in master (no heap allocations or virtual function calls in inner loops).

closed 2022-04-18 by xuchenhan-tri