RobotLocomotion/drake #14341: pydrake: InputPort<Expression>.FixValue() expects np.ndarrays of float, but should expect object

RussTedrake, 2020-11-17:
Per discussion with @hongkai-dai, this is related to https://stackoverflow.com/questions/64617377/typeerror-set-value-incompatible-function-arguments ... where apparently the OP was actually trying to use `AutoDiffXd` (which explains why the repro in `float` didn't help us).

Here is my simple repro:
```
import pydrake.all

plant = pydrake.multibody.plant.MultibodyPlant(0)
file_name = pydrake.common.FindResourceOrThrow(
    "drake/examples/multibody/cart_pole/cart_pole.sdf")
pydrake.multibody.parsing.Parser(plant).AddModelFromFile(file_name)
plant.Finalize()

def MakeVectorVariables(name, dim):
    return np.array([pydrake.symbolic.Variable(f"{name}{i}") for i in range(dim)])

sym_plant = plant.ToSymbolic()
tau = MakeVectorVariables("u", 2)
sym_context = sym_plant.CreateDefaultContext()
sym_plant.get_actuation_input_port(sym_plant.GetModelInstanceByName("iiwa14")).FixValue(sym_context, tau)
```
which results in 
```
<ipython-input-14-acc1fa28c32f> in <module>()
     19 tau = MakeVectorVariables("u", 2)
     20 sym_context = sym_plant.CreateDefaultContext()
---> 21 sym_plant.get_actuation_input_port(sym_plant.GetModelInstanceByName("iiwa14")).FixValue(sym_context, tau)
     22 sym_plant.SetPositions(sym_context, q)
     23 sym_plant.SetVelocities(sym_context, v)
TypeError: set_value(): incompatible function arguments. The following argument types are supported:
    1. (self: pydrake.common.value.Value[BasicVector_[Expression]], arg0: pydrake.systems.framework.BasicVector_[Expression]) -> None
    2. (self: pydrake.common.value.Value[BasicVector_[Expression]], value: pydrake.symbolic.Expression) -> None
    3. (self: pydrake.common.value.Value[BasicVector_[Expression]], value: numpy.ndarray[numpy.float64[m, 1]]) -> None
Invoked with: <pydrake.common.value.Value[BasicVector_[Expression]] object at 0x7f35726ebf70>, array([Variable('u0', Continuous), Variable('u1', Continuous)], dtype=object)
```

I think 3 should read `value: numpy.ndarray[object[m, 1]]`, not `float`.

By contrast, using 
```
q = MakeVectorVariables("q", 2)
v = MakeVectorVariables("v", 2)
sym_plant.get_actuation_input_port(sym_plant.GetModelInstanceByName("iiwa14")).FixValue(sym_context, pydrake.systems.framework.BasicVector_[pydrake.symbolic.Expression](tau))
sym_plant.SetPositions(sym_context, q)
sym_plant.SetVelocities(sym_context, v)
print(sym_context)
```
works.  And if I inspect the `SetPositions` variant, I see
```
TypeError: SetPositions(): incompatible function arguments. The following argument types are supported:
    1. (self: pydrake.multibody.plant.MultibodyPlant_[Expression], context: pydrake.systems.framework.Context_[Expression], q: numpy.ndarray[object[m, 1]]) -> None
    2. (self: pydrake.multibody.plant.MultibodyPlant_[Expression], context: pydrake.systems.framework.Context_[Expression], model_instance: pydrake.multibody.tree.ModelInstanceIndex, q: numpy.ndarray[object[m, 1]]) -> None
```

jwnimmer-tri, 2020-11-17:
The `Xd` here is the source of the problem:

https://github.com/RobotLocomotion/drake/blob/041f6973ed1384c64282892f3754629fc0156607/bindings/pydrake/systems/framework_py_values.cc#L135-L137

I suspect that `X<T>` would work fine.

RussTedrake, 2020-11-17:
awesome.  testing that now...

EricCousineau-TRI, 2020-11-17:
Woot! Will reassign since the PR looks like it's going to merge. Feel free to reassign me if any more work is needed.

PR #14343 (2020-11-17): pydrake: Fix type bug in BasicVector::set_value for non-float types
Resolves #14341

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14343)
commit bb5f8ea1 (2020-11-17): pydrake: Fix type bug in BasicVector::set_value for non-float types (#14343)

Resolves #14341
closed 2020-11-17 by EricCousineau-TRI