RobotLocomotion/drake #14344: multibody: Joint default values don't survive scalar conversion (e.g. `ToAutoDIffXd`)

RussTedrake, 2020-11-17:
```
import pydrake.all

plant = pydrake.multibody.plant.MultibodyPlant(time_step=0.002)
filename = pydrake.common.FindResourceOrThrow(
    "drake/manipulation/models/iiwa_description/urdf/iiwa14_no_collision.urdf")

parser = pydrake.multibody.parsing.Parser(plant)
iiwa = parser.AddModelFromFile(filename)
plant.WeldFrames(plant.world_frame(), plant.GetFrameByName("base"))
plant.Finalize()

# Set default positions:
q0 = [0.0, 0.1, 0, -1.2, 0, 1.6, 0]
index = 0
for joint_index in plant.GetJointIndices(iiwa):
    joint = plant.get_mutable_joint(joint_index)
    if isinstance(joint, pydrake.multibody.tree.RevoluteJoint):
        joint.set_default_angle(q0[index])
        index += 1

print(plant.CreateDefaultContext())

print(plant.ToAutoDiffXd().CreateDefaultContext())
```

results in 
```
::_ Context
------------
Time: 0
States:
  1 discrete state groups with
     14 states
          0  0.1    0 -1.2    0  1.6    0    0    0    0    0    0    0    0

...

::_ Context
------------
Time: 0
States:
  1 discrete state groups with
     14 states
       0 0 0 0 0 0 0 0 0 0 0 0 0 0

```

I would expect the autodiff version to have preserved the non-zero default states.

Related to #13065

sherm1, 2020-11-17:
@joemasterjohn can you take a look at this?

jwnimmer-tri, 2020-11-17:
There is already a PR open.

joemasterjohn, 2020-11-17:
Sorry I should have left a comment here. #14347 has the fix for this bug.

jwnimmer-tri, 2020-11-19:
The PR has merged.  Is the remaining work here maybe to hunt down what _other_ tree member data has gotten overlooked during scalar conversion, and/or to refactor the implementations to make it easier to avoid this problem for future changes?

RussTedrake, 2020-11-19:
Using this to track down other offenders is great, of course.  But I do consider my OP resolved, and would be find if this issue was closed.

joemasterjohn, 2020-11-19:
Since there is another open issue about tree member data not surviving scalar conversion (#14346), I'd advocate for closing this issue and dealing with it there.

closed 2020-11-19 by RussTedrake