RobotLocomotion/drake #14346: multibody: GetBodiesWeldedTo doesn't survive scalar conversion

EricCousineau-TRI, 2020-11-17:
In writing #14345, found that `plant_ad.GetBodiesWeldedTo()` fails with something like this:
```
C++ exception with description "Failure at multibody/topology/multibody_graph.cc:240 in FindBodiesWeldedTo(): condition 'body_index.is_valid() && body_index < num_bodies()' failed." thrown in the test body.
```

Prolly just some topology stuff that needs copying?

Assigning @sherm1 for disposition

\cc @amcastro-tri

EricCousineau-TRI, 2020-11-17:
For now, captured it in a test:
https://github.com/RobotLocomotion/drake/blob/3bcaaf5d733abdfe9329baa2e5c10954626ac4ca/multibody/plant/test/multibody_plant_test.cc#L1387-L1393

sherm1, 2020-11-17:
Yikes -- that's scary. @joemasterjohn, can you look at this one also -- more stuff not surviving ToAutoDiff()?

PR #14411 (2020-12-08): fixes missing multibody graph accounting
Fixes #14346. Adds `MultibodyGraph` accounting for bodies and joints that is missed during plant scalar conversion. This accounting is usually done in `MultibodyPlant::AddRigidBody` and `MultibodyPlant::AddJoint`, but those are not called during scalar conversion, they are cloned from the underlying `MultibodyTree`.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14411)
closed 2020-12-08 by joemasterjohn