RobotLocomotion/drake #14354: API for Body::CalcCenterOfMassTranslationalVelocity()

mitiguy, 2020-11-18:
This issue stems from work associated with PR #14297 and issue #14268, i.e., while working to create MultibodyPlant::CalcCenterOfMassTranslationalVelocityInWorld().

The proposed API in body.h which is overridden in rigid_body.h is shown below.
Per f2f with Sherm, input about this API is requested (@sherm1, @joemasterjohn, @amcastro-tri, ...).

```C++
  /// (Advanced) Calculates the translational velocity of Bcm (`this` body's
  /// center of mass) measured in a frame A, expressed in a frame E.
  /// @param[in] context The context contains the state of the model.
  /// @param[in] frame_A The measured-in frame for Bcm's translational velocity.
  /// @param[in] frame_E The expressed-in frame for the returned result.
  /// @retval v_ABcm_E Bcm's translational velocity in frame A, measured in E.
  virtual Vector3<T> CalcCenterOfMassTranslationalVelocity(
      const systems::Context<T>& context,
      const Frame<T>& frame_A,
      const Frame<T>& frame_E) const = 0;
```

Note: The following code in frame.h shows the methods and underlying code used to calculate
**Frame::SpatialVelocity(context, frame_A, frame_E);**
It is worth noting that the desired API alleviates the end-user of the complexity in these calculations.

```C++
  /// Computes and returns the spatial velocity `V_WF` of `this` frame F in the
  /// world frame W as a function of the state of the model stored in `context`.
  /// @note Body::EvalSpatialVelocityInWorld() provides a more efficient way to
  /// obtain the spatial velocity for a body frame.
  SpatialVelocity<T> CalcSpatialVelocityInWorld(
      const systems::Context<T>& context) const {
    const math::RotationMatrix<T>& R_WB =
        body().EvalPoseInWorld(context).rotation();
    const Vector3<T> p_BF_B = CalcPoseInBodyFrame(context).translation();
    const Vector3<T> p_BF_W = R_WB * p_BF_B;
    const SpatialVelocity<T>& V_WB = body().EvalSpatialVelocityInWorld(context);
    const SpatialVelocity<T> V_WF = V_WB.Shift(p_BF_W);
    return V_WF;
  }

  /// Computes and returns the spatial velocity `V_MF_E` of `this` frame F
  /// measured in `frame_M` and expressed in `frame_E` as a function of the
  /// state of the model stored in `context`.
  /// @see CalcSpatialVelocityInWorld().
  SpatialVelocity<T> CalcSpatialVelocity(
      const systems::Context<T>& context,
      const Frame<T>& frame_M, const Frame<T>& frame_E) const {
    const math::RotationMatrix<T> R_WM =
        frame_M.CalcRotationMatrixInWorld(context);
    const Vector3<T> p_MF_M = this->CalcPose(context, frame_M).translation();
    const Vector3<T> p_MF_W = R_WM * p_MF_M;
    const SpatialVelocity<T> V_WM = frame_M.CalcSpatialVelocityInWorld(context);
    const SpatialVelocity<T> V_WF = this->CalcSpatialVelocityInWorld(context);
    // We obtain V_MF from the composition of spatial velocities:
    const SpatialVelocity<T> V_MF_W = V_WF - V_WM.Shift(p_MF_W);
    if (frame_E.index() == FrameIndex(0)) return V_MF_W;
    // If expressed-in frame_E is not the world, perform additional
    // transformation.
    const math::RotationMatrix<T> R_WE =
        frame_E.CalcRotationMatrixInWorld(context);
    return R_WE.inverse() * V_MF_W;
  }
```

amcastro-tri, 2020-11-18:
I don't understand, what is new in this issue that is not discussed in #14268?
Also, wasn't the conclusion in #14268 to implement the simpler signature `Body::CalcCenterOfMassTranslationalVelocityInWorld()`?

sherm1, 2020-11-18:
#14268 is about _MultibodyPlant_ :: CalcCenterOfMass.... Here Paul is proposing additional methods for _Body_. My thought is that we have a clear use case for the _system_ center of mass calculations (both for whole system and model instances), measured and expressed in World, but we don't yet have any use case for the proposed single-body calculations (except in Paul's implementation for the MBPlant methods, which is internal code). 

So my proposal would be to add the API users want now (system-level CoM stuff) and move on to other things we know are needed now. Defer these body-level CoM sugar methods until we know someone will be using them. The argument above that converting measured & expressed-in frames is complicated isn't persuasive unless someone actually wants to do that! 

As long as a Body can report its CoM pos/vel/acc in its local frame everything else can be computed with the existing API, though a user might require some help doing it. The need for additional sugar is unclear at this point.

mitiguy, 2021-08-31:
At this point, it seems that this issue can be reprioritized low enough that it can be closed.  It can be re-opened if necessary.

closed 2021-08-31 by mitiguy