RobotLocomotion/drake #14381: pydrake bazel: How to get a stacktrace in debug mode?

EricCousineau-TRI, 2020-11-21:
@sloretz is trying to debug some stuff here:
https://github.com/sloretz/drake_ros2_demos/pull/3

I thought #14356 would've fixed it, but possibly not.

He's trying to debug, but can't easily find the point in Drake C++ / pybind11 code where things go awry.

For now, he's trying out the `@traced` macro from here: 
https://drake.mit.edu/python_bindings.html#debugging-with-the-python-bindings

I am assuming the our `cmake -DCMAKE_BUILD_TYPE=Debug` command isn't doing anything more special than `-c dbg` for Bazel; but we shall see...

EricCousineau-TRI, 2020-11-21:
See example trying to get a stack trace in #14380 

Commit: b99ff10d6b
Command:
```
cd drake
bazel run -c dbg //:install -- ${PWD}/build
python_path=$(ls -d ~+/build/lib/python*/site-packages)
env -i PYTHONPATH=${python_path} /usr/bin/python3 -c 'import pydrake.common._module_py._testing as m; m.trigger_segfault()'
```

Triggers a segfault. `coredumpctl gdb` doesn't seem to turn up anything useful (says there's no stacktrace).

Running it under GDB also doesn't seem to yield useful symbols:
```
env -i PYTHONPATH=${python_path} gdb --ex run --args /usr/bin/python3 -c 'import pydrake.common._module_py._testing as m; m.trigger_segfault()'
```

EricCousineau-TRI, 2020-11-21:
Using `python3-dbg` doesn't seem to turn up anything more fruitful in terms of Drake.
(Python, however, shows it all)

EricCousineau-TRI, 2020-11-21:
Naively trying out CMake per our docs:
```
cd drake
mkdir build_cmake && cd build_cmake
cmake .. -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_PREFIX=${PWD}/install
```

EricCousineau-TRI, 2020-11-21:
From my comment in Slack:

> The killer combo with debugging is:
> 
> - Use `@traced` (and the no-buffered-output)
> - Use gdb backtrace
> - Also tell Drake to set it's spdlog level: https://drake.mit.edu/pydrake/pydrake.common.html#pydrake.common.set_log_level
> ```from pydrake.common import set_log_level
> set_log_level("trace")```
> Then it should be obvious how things playout

EricCousineau-TRI, 2020-11-21:
I (or someone else ;) should PR that to Drake docs.

EricCousineau-TRI, 2020-11-21:
Efff... I was the one who sprinkled that sauce in :facepalm: 
https://github.com/RobotLocomotion/drake/commit/ea1b198a7cf52e737dde20b43ae95bf7d9ed4844 (#10103)

pathammer, 2023-03-28:
@EricCousineau-TRI  Is there any way to intercept the [Bazel strip option](https://bazel.build/docs/user-manual#strip) which seems a lot more intuitive?

EricCousineau-TRI, 2023-03-28:
Yup, good idea! Perhaps there's a way to extract that from cc toolchain information.
Unfortunately, though, it's unlikely I will have time to focus on it anytime soon.

Just to check, is this something that impacted your workflow?

pathammer, 2023-03-28:
I lost a day trying to build libdrake with debug symbols before seeing this `--no_strip` option on the installer 😥.
I can file an improvement ticket?

EricCousineau-TRI, 2023-03-30:
Ah, sorry to hear that, and yup, that'd be best!

richardrl, 2023-12-16:
> * set_log_level

traced is indeed awesome. I also built with no_strp and I'm getting a stacktrace now.

set_log_level is no longer present in current drake, what is the equivalent of set_log_level("trace") now?

richardrl, 2023-12-16:
> From my comment in Slack:
> 
> > The killer combo with debugging is:
> > 
> > * Use `@traced` and `reexecute_if_unbuffered`
> > * Use gdb backtrace
> > * Also tell Drake to set it's spdlog level: https://drake.mit.edu/pydrake/pydrake.common.html#pydrake.common.set_log_level
> > 
> > ```
> > from pydrake.common import set_log_level
> > set_log_level("trace")
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Then it should be obvious how things playout

traced is indeed awesome. I also built with no_strp and I'm getting a stacktrace now.

set_log_level is no longer present in current drake, what is the equivalent of set_log_level("trace") now?

closed 2024-11-01 by jwnimmer-tri