RobotLocomotion/drake #14399: World body stores NaN parameters in context.

joemasterjohn, 2020-12-01:
Because each body declares inertial parameters based off of their default parameters, the world body declares a default `SpatialInertia` in the context as its inertial parameters:

```
18 numeric parameter groups with
   10 parameters
     nan nan nan nan nan nan nan nan nan nan
```

This has come up a few times as a red herring when trying to track down the reason for MBP throwing an error ([stack overflow question](https://stackoverflow.com/questions/65093710/extra-body-with-nans-in-context-of-a-multibodyplant-causing-error-while-simulati)) and is at the very least concerning looking to users inspecting the context. While the NaNs in the context are not really a bug (world body is handled as a special case in plant/tree and its default inertial parameters are truly NaN), there might be reason to omit them.

The headache with omiting them for the `world_body` (or any other body with uninitialized parameters) is that methods such as `RigidBody::get_mass(context)` and `RigidBody::CalcSpatialInertiaInBodyFrame(context)` will need special logic to detect that it is `world_body` and NOT retrieve values from the context in that case.

cc: @sherm1 @amcastro-tri thoughts?

sherm1, 2020-12-01:
It does seem bad that World's mass properties are parameters though -- doesn't that mean someone could change them? That doesn't seem like a good idea! My inclination would be to special-case World as a body whose mass properties are fixed and cannot be changed via parameters.

amcastro-tri, 2020-12-02:
I agree with @sherm1, we should not be able to specify parameters for the world. Of course there are different ways to accomplish this in software.

An option:
- We do not even declare parameters for the world body.
- Maybe there is a WorldBody class whose overwrites of get_mass() etc simply throw?