RobotLocomotion/drake #14405: Request to parameterize damping for RevoluteJoint

joemasterjohn, 2020-12-02:
There's a request on a [stackoverflow question](https://stackoverflow.com/questions/65113347/how-to-change-the-damping-coefficient-of-a-revolutejoint-in-drake) to be able to change a revolute joint's damping. I'm using this issue to track the implementation of getters/setters for default damping and context parameters for damping.

RussTedrake, 2021-04-01:
ftr -- I just ran into this limitation, too.  +1 from me.

joemasterjohn, 2021-04-01:
This depends on a fix for #14841, which I'm working on now. But once it's finished I'll put this on my plate.

nepfaff, 2024-02-12:
I would love this feature

wrangelvid, 2024-02-12:
+1 from me too.

joemasterjohn, 2024-02-12:
I'm currently working on this, should have a PR up by tomorrow.

PR #20947 (2024-02-29): [multibody] Make joint damping a system parameter
Closes #14405

This PR gives the full system parameter treatment to joint damping. The new APIs:
```
const VectorX<T>& Joint::GetDampingVector(const Context<T>& context) const
void Joint::SetDampingVector(Context<T>* context, const VectorX<T>& context) const
```
are introduced so that all joint types get a damping parameter. In addition, I added sugar for
single-dof joints:
```
const T& RevoluteJoint::GetDamping(const Context<T>& context) const;
void RevoluteJoint::SetDamping(Context<T>* context, const T& damping) const;
```
ditto for `Prismati [... cut]