RobotLocomotion/drake #14407: render_engine_gl: Should clean up PerceptionProperties munging

EricCousineau-TRI, 2020-12-03:
See: https://github.com/RobotLocomotion/drake/pull/14400#pullrequestreview-543012159

SeanCurtis-TRI, 2021-06-29:
@EricCousineau-TRI You created this issue and the PR this is arose from.

I've been looking at both and I'm a bit flummoxed to figure out what "munging" is and what is actually be achieved. Even the link you've included doesn't inform me (and I wrote the text there).

So, as the author of all of these artifacts, I'd ask you to actually articulate what's the problem that needs to be solved.

EricCousineau-TRI, 2021-07-02:
The goal of #14400 was to try to consolidate (or avoid) mutation and the use of `temp_props`, as it was confusing to reason about the mutation om #14396 (as a contributor, not an author). Also, I was concerned about the cost of copying properties if/when more data is ever stored in them (e.g. full textures?).

That being said, time has passed and it hasn't really been an issue, so if this articulation isn't accurate / useful, then we can close this issue?

SeanCurtis-TRI, 2021-12-04:
Closing as having outgrown clarity/purpose.

closed 2021-12-04 by SeanCurtis-TRI