RobotLocomotion/drake #14409: clang-9 produces linker errors w/ libstdc++-8, but will aggressively chooses newer versions of libgcc-8-dev / libstdc++-8-dev

EricCousineau-TRI, 2020-12-04:
@azeey has gcc-8 installed on Ubuntu Bionic (for development on OSRC's Ignition ecosystem).

He was trying to build Drake with `clang-9` on his machine, but found linker errors like the following:
```sh
$ cd drake
$ git log -n 1 --oneline --no-decorate
688a07c23f setup: add double-conversion, sqlite3, and lzma for vtk (#14392)
$ bazel clean --expunge
$ CC=/usr/bin/clang-9 CXX=/usr/bin/clang++-9 bazel build //common:filesystem_test
...
bazel-out/k8-opt/bin/common/_objs/filesystem_test/filesystem_test.pic.o:filesystem_test.cc:function drake::(anonymous namespace)::NamespaceAliasSmokeTest_ExistTest_Test::TestBody(): error:
undefined reference to 'std::filesystem::__cxx11::path::_M_split_cmpts()'
```

We looked through `bazel build --subcommand` output, found the linker parameters, and then reran those commands through `clang-9` using verbose commands:
```
$ clang-9 -v  @bazel-out/k8-opt/bin/common/filesystem_test-2.params
```

He found that clang-9 was using `gcc-8` / `libstd++-8`, rather than `libstdc++-7.5` (which is what my system chose).

He was able to work around this by purging those packages from his system, e.g.:
```
sudo apt remove cpp-8 g++-8 gcc-8 libasan5 libgcc-8-dev libstdc++-8-dev libubsan1
```

I found that I could reproduce his issue with the following minimum set of packages:
```
sudo apt install libstdc++-8-dev
```

What's more, if I only install `libgcc-8-dev`, then it will use gcc-8's directory (`/usr/lib/gcc/x86_64-linux-gnu/8`), but then complain with errors like:
```
In file included from external/spdlog/src/file_sinks.cpp:8:
external/spdlog/include/spdlog/details/null_mutex.h:6:10: fatal error: 'atomic' file not found
#include <atomic>
```

I dunno proper solution. Perhaps just some docs or whatever? Or just knowing the workaround? Or something more prinicipled?

Disclaimer: I haven't checked upstream, and have timed out atm.

\cc @jwnimmer-tri

EricCousineau-TRI, 2020-12-04:
Aside from `g++-8`, we both had the same versions of `clang-9`, `bazel`, `libstdc++-7-dev`, `ld.gold`, etc.

EricCousineau-TRI, 2020-12-04:
Also, just for the record, output of `ldd` and the linker params on working system:
https://gist.github.com/EricCousineau-TRI/f94bedcb9d5fadbb63802c5aa92cc3b9

soonho-tri, 2020-12-04:
I guess https://drakedevelopers.slack.com/archives/C0KDGF4V9/p1575504727031800 might explain the filesystem part of this issue.

EricCousineau-TRI, 2020-12-04:
Lol, full circle one year later :sweat_smile:

jwnimmer-tri, 2020-12-07:
I wonder if specifying `--gcc-toolchain` is enough to select the intended revision.  If so, we should ensure (or at least document) that it must be specified.

jamiesnape, 2020-12-07:
I think we are aiming to upgrade to `clang-10` once we support Big Sur, but I imagine the some issue will appear that is vaguely similar. I think Jeremy is correct. There is certainly a flag for the compiler/linker (or an environment variable for Bazel).

jwnimmer-tri, 2020-12-07:
I think a fair statement of the problem is that our build (or at least, build instructions) are not reproducible.  All versions of clang default to using the latest GCC from the host, so no matter what other clang upgrades we foresee, we should ensure that we're also locking down the GCC toolchain that Clang will pull in.

So, less about `<filesystem>` acutely, and more about the "choose newer".

jamiesnape, 2020-12-07:
> I think a fair statement of the problem is that our build (or at least, build instructions) are not reproducible. All versions of clang default to using the latest GCC from the host, so no matter [...]

> So, less about <filesystem> acutely, and more about the "choose newer".

Exactly.

jwnimmer-tri, 2021-07-24:
The `--gcc-toolchain` could help us swap between some homedir GCC vs the Ubuntu GCC, but it can't disambiguate the Ubuntu versions.  When using a GCC from `/usr`, Clang will always take the newest one.

https://stackoverflow.com/a/41982109/13032709

https://lists.llvm.org/pipermail/cfe-dev/2016-March/047761.html

Setting up a fake prefix to control which GCC gets used by Clang sounds like a lot of work
... so as a fallback, I'd like to fail fast, but doing so without false positives seems tricky
... so as a fallback, I'd like to add documentation, but we don't really have a page anywhere that details how to build with Clang in the first place
... so as a fallback, this issue will serve as the FAQ and we'll call it a day.

closed 2021-07-24 by jwnimmer-tri