RobotLocomotion/drake #14422: models: franka_description isn't part of `show_model` runfiles?

EricCousineau-TRI, 2020-12-08:
I would've expected that `franka_description` to be part of `@drake//bindindgs/pydrake`'s runfiles, b/c I tried the following from Anzu:
```
bazel run @drake//manipulation/util:show_model -- --meshcat=default --find_resource drake/manipulation/models/franka_description/urdf/panda_arm.urdf
```

However, it reported this file was missing.
Using drake @ 78faa37365, when looking at the `{runfiles}` directory, I only see:
```
$ cd .../show_model.runfiles/
$ ls drake/manipulation/models/
iiwa_description  wsg_50_description  ycb
```

I should do a `bazel query` to see what the path is from `pydrake` to any of the `iiwa` data, and then ensure that some equivalent of `@drake//manipulation/models:install_data` is reflected into the `data` deps of `//:drake_shared_library` or `pydrake`.

jwnimmer-tri, 2020-12-08:
In general, is is _not_ true that all robot models should be a data dependency of pydrake nor libdrake.  They must be a part of `//:install` but unless any code in pydrake or libdrake refers to the model, it's not actually a proper use of `data` to declare it as such.

If we want show_model to depend on all of the models, then we add the models to _its_ data, not libdrake's data.

jwnimmer-tri, 2022-05-02:
As best I understand, this is no bug here.  Closing for now.

closed 2022-05-02 by jwnimmer-tri