RobotLocomotion/drake #14423: franka: Mesh textures don't show up in meshcat?

EricCousineau-TRI, 2020-12-08:
Confirmed per Slack comment here:
https://drakedevelopers.slack.com/archives/CMV686ZD4/p1607385364024400

Results on drake @ d258ef193:
```
cd drake
# Terminal 1
bazel run @meshcat_python//:meshcat-server
# Terminal 2
bazel run //tools:drake_visualizer
# Terminal 3
bazel run //manipulation/util:show_model -- --meshcat=default --find_resource drake/manipulation/models/franka_description/urdf/panda_arm.urdf
```

[image] [image]

*Note*: Due to #14422, I quickly hacked it to include franka:
```
$ cd drake
$ git diff -U1
diff --git a/manipulation/util/BUILD.bazel b/manipulation/util/BUILD.bazel
index 44e6daac85..fb51b75008 100644
--- a/manipulation/util/BUILD.bazel
+++ b/manipulation/util/BUILD.bazel
@@ -113,2 +113,5 @@
     srcs = ["show_model.py"],
+    data = [
+        "//manipulation/models/franka_description:models",
+    ],
     deps = [
```

\cc @sherm1 @blandry

sherm1, 2020-12-08:
cc @rpoyner-tri in case you want to look prior to return of the Geometry King

SeanCurtis-TRI, 2020-12-08:
I believe @joemasterjohn has agreed to take oversight of meshcat. (Joe, correct me if my memory is failing me.)

joemasterjohn, 2020-12-08:
I did make a mention of that, but if @rpoyner-tri wants to take a crack at this I don't mind.

RussTedrake, 2021-09-19:
FYI -- https://github.com/RobotLocomotion/drake/pull/15788 resolves this issue (for meshcat c++, not for meshcat-visualizer in python).

SeanCurtis-TRI, 2021-12-04:
Closing this as meshcat-python is an evolutionary dead end.

closed 2021-12-04 by SeanCurtis-TRI