RobotLocomotion/drake #14426: MultibodyTree "there is no" exceptions should list valid values

ggould-tri, 2020-12-08:
All of the MBP by-name accessors have exceptions that say "there is no [thingy] named [mumble] in the model."

That's great and all but the only way for a user faced with this error to find out what you might have gotten wrong is to run a debug build, get a stack trace, find the enclosing call, and log the contents of the model right before the call.

The exception should list the available names to save the user from this hunt.

sherm1, 2020-12-09:
Is the proposal just to append a list of all possible [thingys] that would not have generated an error message?

ggould-tri, 2020-12-09:
That seems to me the most obvious ("you asked for 'ikea_dinera_plate_8in' but the valid models were 'ikea_dinera_plate_8in_0', 'world', 'default', ...").  But I'd be delighted with other alternatives too, so long as they don't require rebuilding in debug mode to find the failure point and add printfs.

PR #17259 (2022-05-24): Add error guidance to Get*ByName methods.
Along with #17255, resolves #14426

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17259)
commit e9a61130 (2022-05-24): Add error guidance to Get*ByName methods (#17259)

Along with #17255, resolves #14426
closed 2022-05-24 by jwnimmer-tri