RobotLocomotion/drake #14445: No generic way to set random joint states

ggould-tri, 2020-12-10:
The setting of joint random distributions currently requires downstream projects to violate GSG guidance "virtual methods are the preferred way of executing different code paths depending on a specific subclass type".

One of the more heavily used convenience methods in our downstream project is called `MaybeSetJointPosition` and consists largely of:
```
bool MaybeSetJointPosition(
    MultibodyPlant<double>* plant,
    const ModelInstanceIndex instance,
    const std::string& joint_type,
    const std::string& joint_name,
    const Expression& value) {
```
...
```
  if (is_prismatic) {
    PrismaticJoint<double>& joint =
        plant->GetMutableJointByName<PrismaticJoint>(joint_name, instance);
    DRAKE_DEMAND(joint.num_positions() == 1);
    joint.set_random_translation_distribution(value);
    return true;
  }
  if (is_revolute) {
    RevoluteJoint<double>& joint =
        plant->GetMutableJointByName<RevoluteJoint>(joint_name, instance);
    DRAKE_DEMAND(joint.num_positions() == 1);
    joint.set_random_angle_distribution(value);
    return true;
  }
```
This abomination works only for two types of one-dof joint.  It exists because neither MBP itself nor Joint provides a way to set the randomness for a joint; one must manually typeswitch and dispatch to a subclass method.

There should be a way to do this operation generically for a Joint, either within MBP (as we do for free bodies eg with `SetFreeBodyRandomPositionDistribution`) or within the base Joint class (as we do for default positions with `set_default_positions`) or both.

(I am filing this issue in order to reference it from some otherwise defective downstream code)

jwnimmer-tri, 2020-12-10:
\CC @joemasterjohn relates to making the Joint API use consistent nouns and verbs #14291, and how we end up spelling parameters (Expression random, double default, Context actual) across the MbT.

sherm1, 2020-12-10:
Assigning to Rico for disposition.

jwnimmer-tri, 2024-08-28:
The Joint base class offers this method: `void set_default_positions(const VectorX<double>& default_positions)`.  That is enough to default-posture a joint generically, without any GSG-questionable casting hijinks.

Note however -- that function sets the _default_ position, not the _random position distribution_ which is what this issue was asking about.  We had previously embraced that workflow of teach MbP how to randomly sample itself by providing it with the statistics we desire, but these days we generally consider it that to have been a Really Bad Move and should not add more code in support of it.

Statistically sampling random conditions is *orders of magnitude simpler* as a layer that sits atop the MbP.  The sampling code chooses all of the sampled values, and then places them into the MbP (either as defaults, or per-context values).  Then, the MbP doesn't need to know anything about how we'd like to take samples.  Trying to denote the sampling inside of the plant only solves very narrow use cases.

Therefore, I'll close this as "won't fix".  The MbP API we need to enable any-joint randomization already exists.

closed 2024-08-28 by jwnimmer-tri