RobotLocomotion/drake #14455: Trajectory as a signal

jc211, 2020-12-15:
I want to use the trajectory class with a circular buffer implementation. Basically I imagine using it so that I can update it constantly and only keep the last x entries. Would you advise I generalize the trajectory class so that it doesn't always store the breaks as a vector. Maybe I could add a template to specify the underlying storage type. Or perhaps I should just copy the whole implementation and make a new one based on a circular buffer?

My use case is that I have a gyro signal coming in at 400Hz and I want to use the trajectory class as a signal and sample using a first order hold. I don't think this class was meant for this kind of use case, but it has everything already inbuilt to be a good signal abstraction.

rpoyner-tri, 2020-12-15:
Assigning to @EricCousineau-TRI for initial assessment and advice.

EricCousineau-TRI, 2020-12-15:
Thanks Rico!

@jc211 Can I ask why you want to do a circular buffer?
I assume it's for performance optimization, but for your use case, have you identified the growth / reassignment of vectors as your primary bottleneck?

Also, which concrete trajectory implementation are you using? Perhaps `PiecewisePolynomial`?
My suggest is to first try storing your polynomial coefficients in a matrix, and just permuting / shifting the rows in a way using reserved-size matrices - the only concern is that you have breaks reserved at the beginning (e.g. when your "buffer" isn't full).

See [`PiecewisePolynomial<T>::setPolynomialMatrixBlock`](https://drake.mit.edu/doxygen_cxx/classdrake_1_1trajectories_1_1_piecewise_polynomial.html#a9a7863e6bf8a3ccb5335563370305add)

To help us the most, can you produce a simplified C++ (or Python? :shrug:) program that can be profiled? e.g. using [`oprofile`](https://oprofile.sourceforge.io/about/), Google Benchmark, or just a program that uses `std::chrono` and prints timing metrics?

\cc @hongkai-dai (in case you have a better suggestion!)

EricCousineau-TRI, 2020-12-15:
> My use case is that I have a gyro signal coming in at 400Hz and I want to use the trajectory class as a signal and sample using a first order hold. 

Ah, this use case seems pretty simple?
e.g. I personally would just implement my own simple zero order hold structure. I did that here (when tinkering with `libfranka`):
https://github.com/EricCousineau-TRI/libfranka/commit/273a21d

How is your trajectory getting consumed downstream? Are you just sampling from it, or are you feeding it to other components that require the trajectory interface?

jc211, 2020-12-15:
Thanks for your input Eric. 

With regards to how the trajectory is consumed:
I have a system that is producing accelerometer data and gyro data at different rates. To synchronize them and feed them to my rotation estimation system, I wanted to make trajectories out of both streams and then resample them down the line to get synced values that would then be fed to my sensor fusion setup. I was hoping to experiment with different interpolation methods that the piecewise polynomial trajectory class already offers. By converting the underlying storage to a circular buffer, my code gets kind of simple since all I have to do is make sure the buffer has enough capacity so the resampler down the line can run well. 

Based on other use cases I've seen, the trajectory class is almost always used  on some finite set of points where the start point and end point are known. Maybe I'm just hijacking this class to become a buffer with interpolation (and extrapolation) functions. If that is the case, it may be better if I just create a sister copy of piecewise trajectory rather than add some template to trigger the behavior.

jc211, 2021-01-04:
I'm closing this for now so it doesn't pollute the issue pool.

closed 2021-01-04 by jc211