RobotLocomotion/drake #14457: a better pattern for retrieving subsystem and context together

rpoyner-tri, 2020-12-16:
(h/t @ggould-tri )

Seen in anzu code, a somewhat common repeated pattern goes like this (more verbose in actual code):

```
system = GetSubsystemByName("blah");
context = GetSubSystemContext(system);
```

This issue asks whether there should be a more ergonomic API for this in Drake itself.

ggould-tri, 2020-12-16:
One alternative to this would be for subsystems to silently accept any enclosing context, so there would be no need to retrieve a subsystem context anyway.

sherm1, 2020-12-16:
Allowing subsystems to take the Diagram context would be convenient, but in some cases could come with a significant performance penalty due to the need to search for the right subcontext. (I'm thinking of trivial grab-it-out-of-the-cache operations like getting a body's location in space.) In that case we would need also to maintain a lower-level API that accepts only the subcontext for use in performance-touchy circumstances. (Conceivably we could supplement Diagram contexts with a clever flattened data structure allowing any sub- or subsubsystem immediate access to its own sub- or subsubcontext.)

Sugar like
`[subsystem, subcontext] = diagram.GetSubsystemByName("blah", diagram_context);`
could be an improvement though.

jwnimmer-tri, 2020-12-16:
I'm going to wait to offer my thoughts until we have links to three extant code blocks, in situ, that are suffering.  Absent that, we are likely to go off on a tangent.

rpoyner-tri, 2020-12-16:
A little data from anzu:
```
rico@Puget-161804-10:~/checkout/anzu$ git grep -n10 GetSubsystemByName |grep GetSubsystemContext |wc -l
10
```

ggould-tri, 2020-12-16:
Goes up to 16 when you include GetMutableSubsystemContext

sherm1, 2020-12-16:
Related: GetMyContextFromRoot() and GetMyMutableContextFromRoot()

jwnimmer-tri, 2024-11-01:
> Related: GetMyContextFromRoot() and GetMyMutableContextFromRoot()

Yes.  Use that instead.

closed 2024-11-01 by jwnimmer-tri