RobotLocomotion/drake #14464: reusable infrastructure for benchmark measurements

rpoyner-tri, 2020-12-16:
Development of the `cassie_bench` benchmark involved adding a fair amount of infrastructure to control experiment conditions, manage result data, etc. Most of this would be useful for benchmarks and performance-improvement campaigns elsewhere in Drake.

This issue proposes to repackage that infrastructure for reuse. A prerequisite might be to show 3 or so examples where it would be useful. Follow-up would include outfitting those examples with the reusable machinery.

rpoyner-tri, 2020-12-16:
Related tickets: #8482, #14449. There are a few other google_benchmark programs lurking around as well.

rpoyner-tri, 2020-12-16:
https://github.com/rpoyner-tri/drake/tree/performance-tools has some work in progress.

DamrongGuoy, 2021-01-06:
This would be very useful for hydroelastics too. Thank you!

rpoyner-tri, 2021-02-14:
The main reuse PR has merged. I'll keep this open to track at least one new client.

Optionally, it would be nice to figure out a convenient way to better manage the compare.py script from googlebenchmark. Right now, I'm using it from bazel-drake/externals, but since that is bazel-managed cache, the file disappears at random times. Running any rule that depends on googlebenchmark will bring it back, but the workflow is hardly optimal.

rpoyner-tri, 2021-04-30:
The prospects for a new client of this infrastructure don't seem imminent, and there is a TODO regarding compare.py usage. I don't think there is much value in holding this ticket open.

closed 2021-04-30 by rpoyner-tri