RobotLocomotion/drake #14477: drake_py_test rules using pydrake fail allow_import_unittest check

ggould-tri, 2020-12-21:
A unit test which imports pydrake transitively inherits an import of unittest via pydrake's import of numpy (`numpy/testing/__init__.py`:10).  It therefore fails the `allow_import_unittest` check.

Of course the caller can work around this by setting the `allow_import_unittest` override, but that defeats the purpose of the check.

ggould-tri, 2020-12-21:
(This issue is to allow me to TODO some otherwise defective code for #13904)

EricCousineau-TRI, 2020-12-30:
Yeah, that's related to how `drake_py_unittest_main.py` is written - it allows users to not need `if __name__ == "__main__"`, but it does weird implicit things. The `import uniitest` shindig is another factoring of that which I believe is excessively conservative.

I have minor stake in this issue (aligning unittests in Drake to be more explicit, like Anzu, but with linting), but i have no bandwidth to tackle it anytime soon. :shrug: 

FWIW I'd label this as "backlog" as it's not really that big of a pain point? Or just close it?

\cc @jwnimmer-tri

ggould-tri, 2021-01-04:
Yeah, the context here is porting code from anzu.

Since I just TODO'ed to this issue in #14478 I have an interest in not closing it -- IMO we shouldn't close issues that back TODOs.  I'm definitely okay with backlogging it if I'm the only person who cares, though.

jwnimmer-tri, 2021-01-04:
As I commented in the PR, unit tests should almost never use `drake_py_test`.  Almost all of them should use `drake_py_unittest`, which does not suffer from this problem.  My vote is that this caveat is so rare as to not even merit a backlog issue.  Steady state, we will only have 1 or 2 uses of `drake_py_test` in the tree at any time.

jwnimmer-tri, 2021-01-05:
Maybe we should pivot this ticket to be "improve the error message for the 'import unittest' diagnostic", to have it provide a link back to the documentation of `drake_py_test` along with better stardoc on that macro explaining that it's only to be used in the 1% rare cases.

ggould-tri, 2021-01-05:
This is also the view I have come to.  Updating the headline.

jwnimmer-tri, 2024-03-17:
I don't know what to say, the suggestion to use `drake_py_unittest` to solve the problem seems as clear as day to me:

https://github.com/RobotLocomotion/drake/blob/master/common/test_utilities/disable_python_unittest/unittest/__init__.py

Maybe next time this comes up, we can try to dig into what was confusing.  For now: "not planned".

closed 2024-03-17 by jwnimmer-tri