RobotLocomotion/drake #14481: `brew 'glib'` (in `/setup/mac/binary_distribution/Brewfile` is forcing python@3.9

RussTedrake, 2020-12-29:
Which then fails, because we try to use 3.8 everywhere else.
(this has been causing github actions to fail on my manip repo CI: https://github.com/RussTedrake/manipulation/runs/1621729167 )

```
+./htmlbook/workflows/ci/catalina/setup:18> sudo tar -xf /tmp/drake-latest-mac.tar.gz
+./htmlbook/workflows/ci/catalina/setup:19> popd
+./htmlbook/workflows/ci/catalina/setup:21> export HOMEBREW_CURL_RETRIES=4
+./htmlbook/workflows/ci/catalina/setup:23> brew bundle '--file=/opt/drake/share/drake/setup/Brewfile' --no-lock
Tapping dreal-deps/ibex
Tapping robotlocomotion/director
Installing cmake
Installing double-conversion
Installing dreal-deps/ibex/ibex@2.7.4
Installing eigen
Using gcc
Installing gflags
Installing glew
==> Upgrading 1 outdated package:
glib 2.66.2_1 -> 2.66.4
==> Upgrading glib 2.66.2_1 -> 2.66.4 
==> Downloading https://homebrew.bintray.com/bottles/python%403.9-3.9.1_3.catalina.bottle.tar.gz
==> Downloading from https://d29vzk4ow07wi7.cloudfront.net/e545eab0c09efe7653095fb7cfa47d186c331369110c263abafa3a4d7d798796?response-content-disposition=attachment%3Bfilename%3D%22python%403.9-3.9.1_3.catalina.bottle.tar.gz%22&Policy=eyJTdGF0ZW1lbnQiOiBbeyJSZXNvdXJjZSI6Imh0dHAqOi8vZDI5dnprNG93MDd3aTcuY2xvdWRmcm9udC5uZXQvZTU0NWVhYjBjMDllZmU3NjUzMDk1ZmI3Y2ZhNDdkMTg2YzMzMTM2OTExMGMyNjNhYmFmYTNhNGQ3ZDc5ODc5Nj9yZXNwb25zZS1jb250ZW50LWRpc3Bvc2l0aW9uPWF0dGFjaG1lbnQlM0JmaWxlbmFtZSUzRCUyMnB5dGhvbiU0MDMuOS0zLjkuMV8zLmNhdGFsaW5hLmJvdHRsZS50YXIuZ3olMjIiLCJDb25kaXRpb24iOnsiRGF0ZUxlc3NUaGFuIjp7IkFXUzpFcG9jaFRpbWUiOjE2MDkyNDcxMTB9LCJJcEFkZHJlc3MiOnsiQVdTOlNvdXJjZUlwIjoiMC4wLjAuMC8wIn19fV19&Signature=dYYMMLeZszHEO-hdbHcGhVlkZRwGXscQNPMsm7LAQeRq5~cZZA1h~fbfdGwNCgBF3ocic2NPaxlnP4ZjXA9Lx-aI9VVh2bu-GwLk-SMfW465ye-UnsQksNTa0xTIlCc~tpYKWsRpnSLZO6RSE1d0Urk11I5kFsevDg-AmW99GKa5rF9poPZH96~bdetRroqEeZl7gW~RlUJ2M5aFjBhP0eSzPhGjKExiSKnfSBzHGRMr7qAkFMzjmJl9J7h6QyMVfQuM7InXjU5LBeroECvFlpKgBT5jFGiP4owYb7NUrCW~kT5GiRIOMMPOWRYbiC03-ao9bgG2GL0W-N7XFGJOoQ__&Key-Pair-Id=APKAIFKFWOMXM2UMTSFA
==> Downloading https://homebrew.bintray.com/bottles/glib-2.66.4.catalina.bottle.tar.gz
==> Downloading from https://d29vzk4ow07wi7.cloudfront.net/65d130d343e12482908e6faa372af3c80dbcd5d02652cab648282a733c5e4f93?response-content-disposition=attachment%3Bfilename%3D%22glib-2.66.4.catalina.bottle.tar.gz%22&Policy=eyJTdGF0ZW1lbnQiOiBbeyJSZXNvdXJjZSI6Imh0dHAqOi8vZDI5dnprNG93MDd3aTcuY2xvdWRmcm9udC5uZXQvNjVkMTMwZDM0M2UxMjQ4MjkwOGU2ZmFhMzcyYWYzYzgwZGJjZDVkMDI2NTJjYWI2NDgyODJhNzMzYzVlNGY5Mz9yZXNwb25zZS1jb250ZW50LWRpc3Bvc2l0aW9uPWF0dGFjaG1lbnQlM0JmaWxlbmFtZSUzRCUyMmdsaWItMi42Ni40LmNhdGFsaW5hLmJvdHRsZS50YXIuZ3olMjIiLCJDb25kaXRpb24iOnsiRGF0ZUxlc3NUaGFuIjp7IkFXUzpFcG9jaFRpbWUiOjE2MDkyNDcxMTF9LCJJcEFkZHJlc3MiOnsiQVdTOlNvdXJjZUlwIjoiMC4wLjAuMC8wIn19fV19&Signature=VIBk~Og2it0AAZhNOtzDc-wTGCF2Bjpdc6xn3vCSrnnaHnnHNSOGmcX45iHuMY41RvHmXHMElkBZAajxJE5sHWogDqqPSKynxLVJKSvVR858lmpJGolvlZFd-FIHINXC~MCLN~YiurpGT1s6ASwVUm854AsSgPWyP71qGlNApgVaNNN4mWh8jTd1JwjSUoWMaJqzvFIqV8Ff~1jJCTWSf6XOTzN3SKav4rS8-A~YzoLhzVfzzSPpUMM~8Quv3gJMWqEWqm7oOOfXL9RHrfOQr9txMgA8qB5cmpvfk8zjbaXIdJYUrUqaNJdQDdp0ZXg0tcz6EHhVkIUVLRzyV4Wr5g__&Key-Pair-Id=APKAIFKFWOMXM2UMTSFA
==> Installing dependencies for glib: python@3.9
==> Installing glib dependency: python@3.9
==> Pouring python@3.9-3.9.1_3.catalina.bottle.tar.gz
Error: The `brew link` step did not complete successfully
Removing: /usr/local/Cellar/libtiff/4.1.0_1... (247 files, 4.2MB)
Removing: /Users/runner/Library/Caches/Homebrew/libtiff--4.1.0_1.catalina.bottle.tar.gz... (1.1MB)
Warning: Skipping node@14: most recent version 14.15.3 not installed
Removing: /Users/runner/Library/Caches/Homebrew/node@14--14.15.1.catalina.bottle.tar.gz... (16.5MB)
Warning: Skipping packer: most recent version 1.6.6 not installed
Removing: /Users/runner/Library/Caches/Homebrew/packer--1.6.5.catalina.bottle.tar.gz... (59.5MB)
Removing: /usr/local/Cellar/python@3.9/3.9.1... (4,510 files, 71.8MB)
Warning: Skipping ruby: most recent version 3.0.0_1 not installed
Removing: /Users/runner/Library/Caches/Homebrew/ruby--2.7.2.catalina.bottle.tar.gz... (9.2MB)
Warning: Skipping swiftlint: most recent version 0.42.0 not installed
Removing: /Users/runner/Library/Caches/Homebrew/swiftlint--0.41.0.catalina.bottle.tar.gz... (3MB)
Warning: Skipping utf8proc: most recent version 2.6.1 not installed
Removing: /Users/runner/Library/Caches/Homebrew/utf8proc--2.6.0.catalina.bottle.tar.gz... (172.9KB)
Warning: Skipping zstd: most recent version 1.4.8 not installed
Removing: /Users/runner/Library/Caches/Homebrew/zstd--1.4.5.catalina.bottle.tar.gz... (1.3MB)
Removing: /Users/runner/Library/Caches/Homebrew/Cask/google-chrome--87.0.4280.88.dmg... (163.9MB)
Removing: /Users/runner/Library/Caches/Homebrew/Cask/firefox--83.0.dmg... (73.0MB)
Removing: /Users/runner/Library/Caches/Homebrew/Cask/microsoft-edge--87.0.664.60.pkg... (176.6MB)
Removing: /Users/runner/Library/Caches/Homebrew/Cask/microsoft-auto-update--4.29.20110804.pkg... (6.8MB)
Pruned 0 symbolic links and 45 directories from /usr/local
+./htmlbook/workflows/ci/catalina/setup:23> brew --cache
+./htmlbook/workflows/ci/catalina/setup:23> rm -rf /tmp/drake-latest-mac.tar.gz /Users/runner/Library/Caches/Homebrew
Error: Process completed with exit code 1.
```

I believe we need to pin the `glib` version in drake until we finish the migration to 3.9?

RussTedrake, 2021-01-03:
After some hammering, this fixed it for me on github mac CI:
https://github.com/RussTedrake/htmlbook/commit/a9396db4718102bca8203956fd2225b1e03d6223#diff-855180916c5c4636ec4c690e32742ecbbc6022156eae18c81e802a7ca85949b7

soonho-tri, 2021-01-03:
FYI, https://github.com/actions/virtual-environments/issues/2322 might be the culprit here.

jamiesnape, 2021-01-04:
@soonho-tri is correct, the conflict is not ours. My preferred fix is https://github.com/snape/RVO2/blob/9ce1894eb754e3844a60ff048783769e28ff73c8/.github/workflows/ci.yml#L22 before any `brew` operations.

jamiesnape, 2021-01-04:
(`python@3.8` and `python@3.9` do not conflict, a regular install by a developer would not see an error.)

RussTedrake, 2021-01-04:
That works for me.  I've updated my use cases, too.  Feel free to close this now (I'm happy that it's documented) if you think that's best.

jamiesnape, 2021-01-04:
I think we can close. Conflicts with our use of Python 3.8 should be minimal as Homebrew have pretty much already moved everything to 3.9 and we install almost no Python code outside of `lib/python3.8/site-packages`.

GitHub Actions, on the other hand, don't seem to care too much what goes on their images (like Travis CI unfortunately), so other conflicts (Python and non-Python) may happen in future.

closed 2021-01-04 by jamiesnape