RobotLocomotion/drake #14493: Highlight existence of versioned binaries in the documentation

stonier, 2021-01-05:
Recently went looking for a versioned binary of drake, but had to do some digging to get there. The documentation at https://drake.mit.edu/from_binary.html only points to the existence of nightly binaries.

The versioned binaries do exist, but under a versioned tag's Asset list.

*Victory Conditions*

* Documentation points users to both the nightly and versioned binaries.

rpoyner-tri, 2021-01-05:
I'll give this a first look, since I did the last release and still have a few release-process-related action items.

rpoyner-tri, 2021-01-07:
Passing this along to @jamiesnape to investigate or reassign.

jamiesnape, 2021-01-07:
True. We only publicize the versions published on GitHub. Relates to #14482 somewhat. I would fix up the docs now, but I should  wait until #13832 progresses another bullet point or two, which will be early next week.

FTR

* https://drake-packages.csail.mit.edu/drake/release/drake-0.25.0-bionic.tar.gz
* https://drake-packages.csail.mit.edu/drake/release/drake-0.25.0-bionic.tar.gz.sha256
* https://drake-packages.csail.mit.edu/drake/release/drake-0.25.0-bionic.tar.gz.sha512
* https://drake-packages.csail.mit.edu/drake/release/drake-0.25.0-focal.tar.gz
* https://drake-packages.csail.mit.edu/drake/release/drake-0.25.0-focal.tar.gz.sha256
* https://drake-packages.csail.mit.edu/drake/release/drake-0.25.0-focal.tar.gz.sha512
* https://drake-packages.csail.mit.edu/drake/release/drake-0.25.0-mac.tar.gz
* https://drake-packages.csail.mit.edu/drake/release/drake-0.25.0-mac.tar.gz.sha256
* https://drake-packages.csail.mit.edu/drake/release/drake-0.25.0-mac.tar.gz.sha512
* https://drake-packages.csail.mit.edu/drake/release/drake-0.25.0-src.tar.gz
* https://drake-packages.csail.mit.edu/drake/release/drake-0.25.0-src.tar.gz.sha256
* https://drake-packages.csail.mit.edu/drake/release/drake-0.25.0-src.tar.gz.sha512
* https://drake-packages.csail.mit.edu/drake/release/drake-0.25.0/setup_drake_colab.py
* https://drake-packages.csail.mit.edu/drake/release/drake-0.25.0/setup_drake_colab.py.sha256
* https://drake-packages.csail.mit.edu/drake/release/drake-0.25.0/setup_drake_colab.py.sha512

jamiesnape, 2021-01-07:
As far as release process goes, I need to document what I do to upload the files (as well as tagging Docker images, building and publishing soon to be apt packages). All the manual tasks that need to be done after the party has started.

jamiesnape, 2021-01-07:
Probably could be mentioned on each release notes page, but I would need to be in the loop earlier until we have everything automated so there are not dead links for a short period.

rpoyner-tri, 2021-01-07:
FWIW @jwnimmer-tri and I had a  brief talk after the 0.25.0 cycle about automating a good chunk of the later steps of the process. No actual plans yet, but some partially baked ideas. Given my understanding of how much work might be in that, could be useful to improve incrementally for a bit longer. Not entirely helpful notes [here.](https://docs.google.com/document/d/1pn6ZC7v0fJdDjKNux1iDUXMzrtr-2ukdymc7En059yA/edit?usp=sharing)

jwnimmer-tri, 2021-01-21:
The discussion here wandered a bit from the goal.  The problem is simple and stated in the top post -- https://drake.mit.edu/from_binary.html only mentions nightlies.  We should just add a "Versioned releases" header prior to it, with links to how & where to find them.