RobotLocomotion/drake #14527: Continuous Mirror to s3 fails without Sphinx

BetsyMcPhail, 2021-01-13:
Failing job: https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-bionic-gcc-bazel-continuous-mirror-to-s3/1844/

Sphinx was removed from the provisioned images, see [Slack conversation](https://drakedevelopers.slack.com/archives/C270MN28G/p1610478253294000)

To fix the error immediately, this job needs to move to the unprovisioned images but it should not need the documentation prerequisites.

BetsyMcPhail, 2021-01-13:
pinging @jwnimmer-tri

jamiesnape, 2021-01-13:
Moved to unprovisioned. I guess there is an edge case where we might want to cache something needed to build documentation, so passing the `--with-doc-only` flag is not necessarily bad, but we should not really need Sphinx, so something is not quite right.

PR #14528 (2021-01-14): sphinx: Allow sphinx-build to be missing during fetch
This allows for `bazel query` to still succeed, even though trying `bazel build` would fail.

Closes #14527.

Also removes a vestigial macOS path for `sphinx-build`.  Since weeks ago, we no longer support sphinx on macOS.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14528)
closed 2021-01-14 by ggould-tri