RobotLocomotion/drake #14532: DrakeVisualizer depends on order of received LCM messages

SeanCurtis-TRI, 2021-01-14:
## Problem

The `DrakeVisualizer` class talks to the `drake_visualizer` application by means of two types of lcm messages: "load" message and a "draw" message. The load message contains the contents of the scene and the draw message contains the poses for all of the frames specified in the load message. For a draw message to be meaningful, a compatible load message must have previously been processed. Load messages cause geometry to be loaded in an unhelpful configuration (they are all posed with identity transforms in the world frame); every load message should be immediately followed by a draw message so that the visualized state of the world makes sense.

However, LCM does not provide guarantees on message arrival (either the order between messages or that a message arrives at all). `DrakeVisualizer` will broadcast a load-draw pair of messages in quick succession if it detects that `SceneGraph`'s state is different from what has last been "loaded".

Generally, this doesn't cause a problem. A common usage for `DrakeVisualizer` is in a simulation context. Even if the initial load-draw pair is out of order (such that the draw comes first), the first message causes a console error for `drake_visualizer`, the load message is handled, and then the simulation dispatches a long sequence of draw messages (typically at 60Hz). So, if the scene is drawn badly because of the out-of-order delivery, it's only for 16 ms.

However, there is a work flow in which this causes problems: the one-off visualization. The original state (or modified state of `SceneGraph`) is simply drawn once. If the messages arrive out of order, there are no subsequent draw messages to rectify the error; there isn't a 16 ms visual artifact, there's simply a useless visualization.

## Solution

Merge the two messages into a single message containing scene and poses. However, we want to minimize the impact on performance. To that end:

1. The scene description is *optional*. It can be omitted from the message. `DrakeVisualizer` uses the current logic to decide whether the data needs to be added or not.
2. `drake_visualizer` detects if the scene description is provided; if so, it rebuilds the scene and poses it according to the poses.

To total overall network traffic remains constant.

jwnimmer-tri, 2021-12-14:
@joemasterjohn  I propose closing this as a"WONTFIX" per #16215. What do you think?

SeanCurtis-TRI, 2021-12-14:
I agree. The circumstance in which it came up was sufficiently rare. Perhaps the best short-term resolution is to document the problem scenario and give some guidance to resolve it.

jwnimmer-tri, 2023-07-30:
The issue of a one-shot visualization going awry does not seem to happen often.

Per the above discussion, let's close this as "won't fix".

closed 2023-07-30 by jwnimmer-tri