RobotLocomotion/drake #14544: drake/manipulation/kinova_jaco does not build

mpatalberta, 2021-01-19:
I would expect a complete build to also do the complete code that is under the manipulation directory.
Thanks for your help

jwnimmer-tri, 2021-01-19:
Can you provide information on exactly what steps you performed, and what seemed to be missing afterwards or go wrong?

A quick skim of the source code, as well as the https://github.com/RobotLocomotion/drake/releases/tag/v0.25.0 precompiled images, shows me that the kinova_jaco libraries should be compiled under normal circumstances.

jwnimmer-tri, 2021-01-21:
I'm going to close this out as "needs more information".  Here's our stock information for how to get help -- https://drake.mit.edu/getting_help.html.

In this case, my best guess is that the software and build system are working as intended, but our documentation or tutorials are lacking or misleading, leading to some unexpected failure to build in your circumstances.  In that case, our preference is to use the community StackOverflow forum a with `#drake` tag to seek assistance in working through the build instructions.  If we find a documentation bug, we can return to the issue tracker here to consolidate it, but there's generally a larger community and better discussion forum on StackOverflow for support questions like this one.

If you are still having difficulty, please re-post this question to StackOverflow with more details.

closed 2021-01-21 by jwnimmer-tri

mpatalberta, 2021-01-21:
> [External Email]
>
> Closed #14544<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_RobotLocomotion_drake_issues_14544&d=DwMCaQ&c=Qwsh1H-X9ypOoLLEcAIltQ&r=s3aTjjowuIOYrRraGZdX6w&m=xGC-6oqTk00wr9TbZAtfi-4EMRXkm3DGBu_w201t7I4&s=yNJT-dwreIA-r0PxmqxsCfEnzRQSB1yW4rTgjcr2WT4&e=>.
>
>
> - CONFIDENTIAL-
>
> This email and any files transmitted with it are confidential, and may also be legally privileged. If you are not the intended recipient, you may not review, use, copy, or distribute this message. If you receive this email in error, please notify the sender immediately by reply email and then delete this email.

Jeremy,
My problem was not understanding what is going on.
The problem was not understanding the use of these files.
Pat.

jwnimmer-tri, 2021-01-21:
Sure thing, no problem.  We are eager to help support new users who have questions, but we'd like to host those discussions on the StackOverflow site, rather than the issues database here.

https://stackoverflow.com/questions/tagged/drake