RobotLocomotion/drake #14552: Periodic update events with small offsets are lost as Context times become large

edrumwri, 2021-01-21:
The periodic event offset is subtracted from the current time here: 

https://github.com/RobotLocomotion/drake/blob/5b3377b92387cd149d5aa7b88f8ce5866347bbc1/systems/framework/leaf_system.cc#L34

in order to determine the next update time for periodic updates. When the Context time is large (> 16 orders of magnitude greater than the offset), this subtraction effectively results in a no-op.

SeanCurtis-TRI, 2021-01-21:
How real a problem is this?

Let's say we have a high frequency period (say 1 ms). We'd cross the threshold you describe when we have 1e13 seconds clocked in the context. 1e13 s / 60 / 60 / 24 / 356 = 325,000 years. And even if we wanted to reduce the period to microseconds, that's still 325 years. Kind of hard to imagine the utility of such a simulation. Would we be simulating a robot that can survive that long?

Obviously, we can't fix fundamental numerical precision. Are you proposing we detect such an event and at least stop the silent no-op?

edrumwri, 2021-01-21:
It doesn't matter how likely you think it is that this will happen. It will fail silently and it's fixable.

SeanCurtis-TRI, 2021-01-21:
I absolutely agree with the stated facts at hand. And while I can easily imagine how someone would encounter the bug, I'm just hard-pressed someone taking that action. But that's neither here nor there.

I'm curious about your "fixable" idea. In my mind, the best we can hope for is detect it and not continue. Better to stop than to lie, after all. Did you have something else in mind?

sherm1, 2021-01-21:
I'd be tempted to include this under a general "Drake uses finite precision to approximate real numbers" warning -- there are lots of potential failures due to limited precision. For example, our measurements of small penetrations between objects A and B become less precise as A and B move further from the origin, eventually becoming nonsense. Do you think this particular case is worse than other problems caused by finite precision?

SeanCurtis-TRI, 2021-01-21:
Or if you allow the bodies/geometries to move far away from the origin.

SeanCurtis-TRI, 2021-01-21:
Certainly, testing `t + Δt != t` in appropriate locations would at least allow us to announce, "Hey! We've got incompatible time values!" instead of just moving on with the illusion of meaningful computations. We do have a reasonable expectation of well-behaved time.

sherm1, 2021-01-21:
We should verify that optimizers are forbidden from cleverly converting the expression `t + Δt != t` to the mathematically (but not numerically) equivalent test `Δt != 0`. I _think_ that's forbidden under normal IEEE rules.

edrumwri, 2021-01-21:
> Do you think this particular case is worse than other problems caused by finite precision?

**It was worse for me** because I want to trigger a system right after another system, periodically. That makes it tricky without consideration of how large the time can get.

That seems a lot worse than times get absurdly large (1e+13s would be needed for a common step size of 1e-3), causing time to no longer advance.

sherm1, 2021-01-21:
One way to fix this would be to have the Simulator refuse to let time go past some arbitrary-but-safe number like 1e10. Do you think that would be acceptable?

SeanCurtis-TRI, 2021-01-21:
In talking to @edrumwri, the sense I got is that he wanted to get two systems to evaluate in a fixed order. So, they both had a periodic event, but one had a *very* small offset so that the state it operated was essentially the state after the previous system concluded. The smaller the offset, the more true this is. But the more likely he is to run into the problem.

For example, if offset is 1e-6, the largest time value he can handle is 1e10 before the offset disappears. However, if he wants a much *tighter* window (e.g., 1e-12), then the time can only run to 1e4, etc. And an offset of epsilon, dies disappears after a second of simulation. So, the more he pushes this towards the logical abstraction of "simultaneous, but ordered events", the more likely he is to run into this wall.

@edrumwri, please correct me where I've got this wrong.

edrumwri, 2021-01-21:
That's correct @SeanCurtis-TRI 

> One way to fix this would be to have the Simulator refuse to let time go past some arbitrary-but-safe number like 1e10. Do you think that would be acceptable?

The problem with this is that the construction of this particular system is far removed from the construction of the Simulator, i.e., it would be quite brittle.

Given the engineering effort required to address this problem correctly, I'd be happy with just a DRAKE_DEMAND that `offset_time` has actually changed when `offset != 0` in the permalink.