RobotLocomotion/drake #14555: Test on new Apple arm hardware

jwnimmer-tri, 2021-01-21:
New Apple laptops ship with arm (not x86) processors.  There is an emulation layer (Rosetta 2) for x86 binaries to continue to run.

We should experiment and test how Drake behaves on this new hardware, at minimum so we can inform users what to expect, but ideally to inform a work plan to patch up any problems, so that at least _some_ installation method can work for users who want to run Drake on this platform.

Any work related to _native_ code support (arm), will be in a separate issue.

jwnimmer-tri, 2021-12-08:
Some additional context:

The goal of this issue is to report on the landscape of what is or isn't working / what the work plan will need to be for Drake to add official support for that hardware.  There is no need to submit patches to _fix_ a broken Drake on these platforms.  This is a "survey" ticket, not a "fix" ticket.

One important question is whether Drake's https://drake-packages.csail.mit.edu/drake/nightly/drake-latest-mac.tar.gz pre-compiled binary will run unchanged. 

It's also important to learn whether we can build on this platform from source.  (Building in x86_64 mode, i.e., still using Rosetta while running.)

Both are useful to know, and I don't have any request for which of the two questions you poke at first.

jwnimmer-tri, 2021-12-13:
For reference, #13514 is about ARM support (which is distinct from this issue's investigation of Rosetta's capabilities).

mwoehlke-kitware, 2021-12-22:
`drake-visualizer` from the precompiled package starts up okay. What else would constitute a good functionality test?

mwoehlke-kitware, 2021-12-22:
```
$ bazel test //...
INFO: Analyzed 8612 targets (0 packages loaded, 0 targets configured).
INFO: Found 3450 targets and 5162 test targets...
FAIL: //multibody/tree:spatial_inertia_test (see /private/var/tmp/_bazel_USER/db4fc5c80bb85819057f46c09c9ed04c/execroot/drake/bazel-out/darwin-opt/testlogs/multibody/tree/spatial_inertia_test/test.log)
INFO: From Testing //multibody/tree:spatial_inertia_test:
==================== Test output for //multibody/tree:spatial_inertia_test:
Using drake_cc_googletest_main.cc

[==========] Running 18 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 18 tests from SpatialInertia
[ RUN      ] SpatialInertia.DefaultConstructor
[       OK ] SpatialInertia.DefaultConstructor (0 ms)
[ RUN      ] SpatialInertia.ConstructionFromMasComAndUnitInertia
[       OK ] SpatialInertia.ConstructionFromMasComAndUnitInertia (0 ms)
[ RUN      ] SpatialInertia.CastToAutoDiff
[       OK ] SpatialInertia.CastToAutoDiff (0 ms)
[ RUN      ] SpatialInertia.ShiftOperator
[       OK ] SpatialInertia.ShiftOperator (0 ms)
[ RUN      ] SpatialInertia.PlusEqualOperator
[       OK ] SpatialInertia.PlusEqualOperator (0 ms)
[ RUN      ] SpatialInertia.ReExpress
[       OK ] SpatialInertia.ReExpress (0 ms)
[ RUN      ] SpatialInertia.Shift
[       OK ] SpatialInertia.Shift (0 ms)
[ RUN      ] SpatialInertia.IsPhysicallyValidWithZeroMass
[       OK ] SpatialInertia.IsPhysicallyValidWithZeroMass (0 ms)
[ RUN      ] SpatialInertia.IsPhysicallyValidWithBadInertia
multibody/tree/test/spatial_inertia_test.cc:356: Failure
matcher(err.what(), expected_msg) evaluates to false, where
err.what() evaluates to "Spatial inertia fails SpatialInertia::IsPhysicallyValid().\n mass = 1\n Center of mass = [0  0  0]\n Inertia about point P, I_BP =\n[  -2, -0.1, -0.2]\n[-0.1,   -3, -0.3]\n[-0.2, -0.3,   -4]\n Principal moments of inertia about Bcm (center of mass) =\n[-4.105976670111365  -2.9188291125626042  -1.9751942173260322]\n"
expected_msg evaluates to "Spatial inertia fails SpatialInertia::IsPhysicallyValid\\(\\).\n mass = 1.0\n Center of mass = \\[0.0  0.0  0.0\\]\n Inertia about point P, I_BP =\n\\[  -2, -0.1, -0.2\\]\n\\[-0.1,   -3, -0.3\\]\n\\[-0.2, -0.3,   -4\\]\n Principal moments of inertia about Bcm \\(center of mass\\) =\n\\[-4.105976670111\\d+  -2.9188291125626\\d+  -1.9751942173260\\d+\\]\n"
[  FAILED  ] SpatialInertia.IsPhysicallyValidWithBadInertia (0 ms)
[ RUN      ] SpatialInertia.IsPhysicallyValidWithCOMTooFarOut
multibody/tree/test/spatial_inertia_test.cc:379: Failure
matcher(err.what(), expected_msg) evaluates to false, where
err.what() evaluates to "Spatial inertia fails SpatialInertia::IsPhysicallyValid().\n mass = 1\n Center of mass = [2  0  0]\n Inertia about point P, I_BP =\n[0.4,   0,   0]\n[  0, 0.4,   0]\n[  0,   0, 0.4]\n Inertia about center of mass, I_BBcm =\n[ 0.4,    0,    0]\n[   0, -3.6,    0]\n[   0,    0, -3.6]\n Principal moments of inertia about Bcm (center of mass) =\n[-3.6  -3.6  0.4]\n"
expected_msg evaluates to "Spatial inertia fails SpatialInertia::IsPhysicallyValid\\(\\).\n mass = 1.0\n Center of mass = \\[2.0  0.0  0.0\\]\n Inertia about point P, I_BP =\n\\[0.4,   0,   0\\]\n\\[  0, 0.4,   0\\]\n\\[  0,   0, 0.4\\]\n Inertia about center of mass, I_BBcm =\n\\[ 0.4,    0,    0\\]\n\\[   0, -3.6,    0\\]\n\\[   0,    0, -3.6\\]\n Principal moments of inertia about Bcm \\(center of mass\\) =\n\\[-3.6  -3.6  0.4\\]\n"
[  FAILED  ] SpatialInertia.IsPhysicallyValidWithCOMTooFarOut (0 ms)
[ RUN      ] SpatialInertia.IsPhysicallyValidThrowsNiceExceptionMessage
multibody/tree/test/spatial_inertia_test.cc:425: Failure
matcher(err.what(), expected_msg) evaluates to false, where
err.what() evaluates to "Spatial inertia fails SpatialInertia::IsPhysicallyValid().\n mass = 0.634\n Center of mass = [0  0.016  -0.02]\n Inertia about point P, I_BP =\n[  0.0023989,    0.000245,     1.3e-05]\n[   0.000245,   0.0023566,  0.00020438]\n[    1.3e-05,  0.00020438, 0.000570304]\n Inertia about center of mass, I_BBcm =\n[0.001983, 0.000245,  1.3e-05]\n[0.000245, 0.002103,  1.5e-06]\n[ 1.3e-05,  1.5e-06, 0.000408]\n Principal moments of inertia about Bcm (center of mass) =\n[0.0004078925412357755  0.0017908225928030743  0.002295284865961151]\n"
expected_msg evaluates to "Spatial inertia fails SpatialInertia::IsPhysicallyValid\\(\\).\n mass = 0.634\n Center of mass = \\[0.0  0.016  -0.02\\]\n Inertia about point P, I_BP =\n\\[  0.0023989,    0.000245,     1.3e-05\\]\n\\[   0.000245,   0.0023566,  0.00020438\\]\n\\[    1.3e-05,  0.00020438, 0.000570304\\]\n Inertia about center of mass, I_BBcm =\n\\[0.001983, 0.000245,  1.3e-05\\]\n\\[0.000245, 0.002103,  1.5e-06\\]\n\\[ 1.3e-05,  1.5e-06, 0.000408\\]\n Principal moments of inertia about Bcm \\(center of mass\\) =\n\\[0.0004078925412\\d+  0.001790822592803\\d+  0.00229528486596\\d+\\]\n"
[  FAILED  ] SpatialInertia.IsPhysicallyValidThrowsNiceExceptionMessage (0 ms)
[ RUN      ] SpatialInertia.SkipValidityCheck
[       OK ] SpatialInertia.SkipValidityCheck (0 ms)
[ RUN      ] SpatialInertia.MakeFromCentralInertia
[       OK ] SpatialInertia.MakeFromCentralInertia (0 ms)
[ RUN      ] SpatialInertia.KineticEnergy
[       OK ] SpatialInertia.KineticEnergy (0 ms)
[ RUN      ] SpatialInertia.MultiplyByEigenMatrix
[       OK ] SpatialInertia.MultiplyByEigenMatrix (0 ms)
[ RUN      ] SpatialInertia.SymbolicNan
[       OK ] SpatialInertia.SymbolicNan (0 ms)
[ RUN      ] SpatialInertia.SymbolicConstant
[... cut ...]
  /private/var/tmp/_bazel_USER/db4fc5c80bb85819057f46c09c9ed04c/execroot/drake/bazel-out/darwin-opt/testlogs/examples/pendulum/trajectory_optimization_simulation_test/test_attempts/attempt_1.log
  /private/var/tmp/_bazel_USER/db4fc5c80bb85819057f46c09c9ed04c/execroot/drake/bazel-out/darwin-opt/testlogs/examples/pendulum/trajectory_optimization_simulation_test/test_attempts/attempt_2.log

Executed 3 out of 5162 tests: 5160 tests pass and 2 fail locally.
There were tests whose specified size is too big. Use the --test_verbose_timeout_warnings command line option to see which ones these INFO: Build completed, 2 tests FAILED, 4 total actions
```

sherm1, 2021-12-22:
Oh, that's more over-sensitivity to number formatting as in #16188.
cc @mitiguy

mwoehlke-kitware, 2022-01-11:
BTW, to test/use this:
- Use `arch -x86_64 /bin/bash` to obtain a shell.
- Make sure there is not an ARM version of brew (`/opt/homebrew/bin/brew`) in your `PATH`.
- Install as usual *from the "x86" shell*.

jwnimmer-tri, 2022-01-13:
The remaining work here is to:
(1) weave this into the user documentation
(2) file follow-up tickets for next steps (e.g., adding apple silicon CI)

PR #16381 (2022-01-14): [doc] Explain how to run Drake under Rosetta 2
Closes #14555.

~This PR also contains a small unrelated docs fixup for today's release notes.~

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16381)
closed 2022-01-14 by jwnimmer-tri