RobotLocomotion/drake #14556: kinova arm with j2s6s300 and drake

mpatalberta, 2021-01-21:
How do you build the urdf files from the kinova arm j2s6s300_standalone.xacro file.
I tried running ros xacro to urdf but that did not work.
I get an error:
[2021-01-21 14:04:19.196] [console] [warning] Currently MultibodyPlant does not handle joint limits for continuous models. However some joints do specify limits. Consider setting a non-zero time step in the MultibodyPlant constructor; this will put MultibodyPlant in discrete-time mode, which does support joint limits.
[2021-01-21 14:04:19.196] [console] [warning] Joints that specify limits are: `j2s6s300_joint_1`, `j2s6s300_joint_2`, `j2s6s300_joint_3`, `j2s6s300_joint_4`, `j2s6s300_joint_5`, `j2s6s300_joint_6`, `j2s6s300_joint_finger_1`, `j2s6s300_joint_finger_tip_1`, `j2s6s300_joint_finger_2`, `j2s6s300_joint_finger_tip_2`, `j2s6s300_joint_finger_3`, `j2s6s300_joint_finger_tip_3`.
terminate called after throwing an instance of 'std::runtime_error'
  what():  This multibody tree already has a mobilizer connecting these two bodies. More than one mobilizer between two bodies is not allowed
Aborted (core dumped)

rpoyner-tri, 2021-01-21:
Hi Patrick -- can you give us more detail on the file you are using and the steps you took to try to load it in Drake? A quick internet search did not locate the model file.

Bouncing to @sherm1 for proper delegation.

jwnimmer-tri, 2021-01-21:
A likely source of `j2s6s300_standalone.xacro` would be https://github.com/Kinovarobotics/kinova-ros/tree/master/kinova_description/urdf.

mpatalberta, 2021-01-21:
Hello,
I have a kinova j2s6s300 robot arm and hand. Also referred to JACO2 .
I see in the manupilation/models/jaco_description/urdf
  You deal with urdf file.
  The files from Kinova come in xacro format.
  Drake does not appear to support xacro files.
  You can however convert a xacro fileto a urdf file by using ROS xacro.
  It is not clear to me how you would go about building the hand,arm and collision file from the converted urdf file.
  Can you give any guidance as to how to resolve this issue.
  I could checkin the files have added .
Pat,

[External Email]

- CONFIDENTIAL-

This email and any files transmitted with it are confidential, and may also be legally privileged. If you are not the intended recipient, you may not review, use, copy, or distribute this message. If you receive this email in error, please notify the sender immediately by reply email and then delete this email.

mpatalberta, 2021-01-21:
Hello,
I had already added and converted the files

j2s6s300_standalone.xacro  and j2s6s300.xacro

I converted them both to a single urdf file using ROS xarco to urdf.
This produces 1 file.
j2s6s300.urdf
problem how did someone generate

  1.  Collision file
  2.  Hand
  3.  Arm
Thanks pat,
Pat,

[External Email]

rpoyner-tri, 2021-01-21:
It appears that Drake has a batch of converted urdf models for the 7DOF variant. All appear to have been hand-edited.

Files are here: https://github.com/RobotLocomotion/drake/tree/master/manipulation/models/jaco_description

cc: @sammy-tri who seems to have converted the 7DOF models, for advice.

sammy-tri, 2021-01-21:
> It appears that Drake has a batch of converted urdf models for the 7DOF variant. All appear to have been hand-edited.

This is correct, particularly with regard to the  `*_arm.urdf`,  `*_hand.urdf`, and  `*_sphere_collision.urdf` models of the 7-dof Jaco.  I created them by hand in #13007.  Splitting out the arm-only and hand-only models was a fairly straightforward copy/paste from the combined model in `j2s7s300.urdf`.  Creating the sphere collision models was rather more time consuming.  To the extent that the 6-dof Jaco2 has a number of links with the same geometry, you could probably copy the `<collision>` elements for those links into a 6-dof version.  It would certainly be more convenient if we had `xacro` or something available to autogenerate these, but alas we do not.  Depending on your application, you might not need these versions of the files at all.

Are you still having issues loading your new j2s6s300.urdf model into `MultibodyPlant`?

mpatalberta, 2021-01-21:
Rick.
My name Patrick Brown.
I have j2s6s300 ( 6dof sphereical with 3 fingers)
Kinova supplied the xacro files.
   J2s6s300_standalone.xacro and j2s6s300.xacro [ standalone includes j2s6s300.xacro.
The drake system does not appear to support xacro.

Ros has a package called xacro.
I rand the program xacro from ros.

My problem here is what was done is not really documented.
Running the suggested manipulation/util/geometry_inspector does not pass for the j2s7s300.xacro or any others.
Further it is not clear where the collision matrix comes.

Help here would be appreciated.

Pat,

[External Email]

sammy-tri, 2021-01-21:
oh, also, to confirm, `j2s7s300.urdf` was created using xacro from ROS and then added to drake.

mpatalberta, 2021-01-21:
My Creasey,
My response is below.

You get from moveit in ROS you get an srdf file that is the collision matrix. Could that be used?

Do I not always need these files in order for the kinematics to work?

Yes I get the error about multibodyplant. Do you know what the error means.

mpatalberta, 2021-01-21:
Mr. Creasy,
I kind guessed that.
Thanks,
Patrick,

[External Email]

sammy-tri, 2021-01-21:
>  You get from moveit in ROS you get an srdf file that is the collision matrix. Could that be used?

I'm not very familiar with the SRDF format or MoveIt!, but I don't think it would be useful to build a collision model for use in drake.   Collision/contact in drake generally works best with a collision model made from geometric primitives instead of meshes, which is why I hand-made the sphere collision model for the 7DOF robot.

> > Depending on your application, you might not need these versions of the files at all

> Do I not always need these files in order for the kinematics to work?
The `j2s6s300.urdf` generated from xacro should be sufficient for kinematics.  Some degree of motion planning and simulation is also possible without a collision model, though there would be limitations.

The error which you're seeing from `MultibodyPlant` usually means that more than one joint has been added to the model between the same pair of bodies (which isn't supported in `MultibodyPlant`).  Unfortunately the error message doesn't say which bodies have the duplicate joint associated with them.  Can you post a link to the urdf file which you generated?  That would make it easier to determine if the duplicate joint is coming from the URDF or being added to the `MultibodyPlant somewhere else.

mpatalberta, 2021-01-21:
Mr. Creasey,

I wish I was not familiar moveit also.

I have put all my files the fork and branch that I have created.

drake/manipulation/models/jaco_discription/urdf
   the input xacro and output urdf are included.

Patrick,

[External Email]

sammy-tri, 2021-01-21:
Looking at the newly generated URDF, it looks like it has a joint which connects the base to the world inside the URDF: https://github.com/mpatalberta/drake/blob/kinova_j2s6s300_arm/manipulation/models/jaco_description/urdf/j2s6s300.urdf#L47

Our arm models in drake don't usually have a joint like this in the URDF file, so I'm guessing that the joint is being added twice, first when the URDF is parsed, and again later in code while constructing the rest of the `MultibodyPlant`.  If you remove the `connect_root_and_world` joint from the URDF, does that help?

mpatalberta, 2021-01-21:
Mr. Creasey,
Thanks for your help I will try the change.
Patrick

[External Email]

mpatalberta, 2021-01-23:
Mr Creasey,
I am having bulid a urdf file from scratch as the Drake system is very weak on this.
They had file for a j2s7s300 wiith collision.
The joint sizes are mm:
275.2
205
205
207.3
103.8
103.8
160

E2 = (a little offset) 0.0098meter
Centerline offset to middle of palm 13.3mm

The j2s6s300 has 1 less joint and is the same as the j2s7s300
The 205 + 205 = 410mm

My plan is to combine these joints to support the j2s6s300
Do you know how I am to merge the collision records for these two joints.
I have highlighted in the joints that are to be merged.
<linkname="j2s7s300_link_3">
    <visual>
      <geometry>
        <mesh filename="../meshes/arm_half_2.obj"/>
      </geometry>
      <material name="carbon_fiber">
        <color rgba="0.792156862745098 0.819607843137255 0.933333333333333 1"/>
      </material>
    </visual>
    <visual>
      <geometry>
        <mesh filename="../meshes/ring_big.obj"/>
      </geometry>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0.0 0.028 -0.232"/>
      <geometry>
        <sphere radius="0.015"/>
      </geometry>
    </collision>
    <collision>
      <origin rpy="0 0 0" xyz="0.03 0.03 -0.20"/>
      <geometry>
        <sphere radius="0.015"/>
      </geometry>
    </collision>
    <collision>
      <origin rpy="0 0 0" xyz="-0.03 0.03 -0.20"/>
      <geometry>
        <sphere radius="0.015"/>
      </geometry>
    </collision>
    <collision>
      <origin rpy="0 0 0" xyz="0.025 0.025 -0.13"/>
      <geometry>
        <sphere radius="0.015"/>
      </geometry>
    </collision>
    <collision>
      <origin rpy="0 0 0" xyz="-0.025 0.025 -0.13"/>
      <geometry>
        <sphere radius="0.015"/>
      </geometry>
    </collision>
    <collision>
      <origin rpy="0 0 0" xyz="0.02 0.02 -0.11"/>
      <geometry>
        <sphere radius="0.02"/>
      </geometry>
[... 178 lines cut ...]
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0.0 -0.00"/>
      <geometry>
        <sphere radius="0.035"/>
      </geometry>
    </collision>
    <collision>
      <origin rpy="0 0 0" xyz="0 0.0 -0.025"/>
      <geometry>
        <sphere radius="0.0325"/>
      </geometry>
    </collision>
    <collision>
      <origin rpy="0 0 0" xyz="0 0.02 -0.105"/>
      <geometry>
        <sphere radius="0.035"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="0.463"/>
      <origin xyz="0 0.0028848942 -0.0541932613"/>
      <inertia ixx="0.0004321316048" ixy="0" ixz="0" iyy="0.0004321316048" iyz="0" izz="9.26e-05"/>
    </inertial>
  </link>

mpatalberta, 2021-01-24:
Mr. Creasey,

My plan is to use the supplied js2s7300_spherical_colllsion.urdf and remove joint 4 /link 4 and then double the length of joint 3/link 3.

Then rename link/joint 5 ,link/joint 6 , link/joint  7 to 4 , 5 and 6 respectably.
Page 44
https://www.kinovarobotics.com/sites/default/files/UG-009_KINOVA_Gen2_Ultra_lightweight_robot_User_guide_EN_R02.pdf
The area above describes the 2 arms and hands we are using here.

As a side note do you have Kinova arm and hand?
The following code is used to talk to my j2s6s300 arm and hand.
Unless you have another suggestion.
https://github.com/mpatalberta/drake-jaco-driver

Thanks,

Patrick Brown

mpatalberta, 2021-01-25:
Mr. Creasley,
I have uploaded my l2s6s300_sphere_collision.urdf file.
Using the drake visualizer it looks correct.
I change the example code to run this arm but the results are still not correct.
Could you please examine change and suggest what othr changes are required,
Thanks,
Patrick Brown
Zebra technology
631 880 1188

From: Brown, Patrick<mailto:Patrick.Brown@zebra.com>
Sent: Sunday, January 24, 2021 2:20 PM
To: RobotLocomotion/drake<mailto:reply@reply.github.com>
Cc: Beach, Robert<mailto:Bob.Beach@zebra.com>
Subject: Re: [RobotLocomotion/drake] kinova arm with j2s6s300 and drake (#14556)

Mr. Creasey,

My plan is to use the supplied js2s7300_spherical_colllsion.urdf and remove joint 4 /link 4 and then double the length of joint 3/link 3.

Then rename link/joint 5 ,link/joint 6 , link/joint  7 to 4 , 5 and 6 respectably.
Page 44
https://www.kinovarobotics.com/sites/default/files/UG-009_KINOVA_Gen2_Ultra_lightweight_robot_User_guide_EN_R02.pdf
The area above describes the 2 arms and hands we are using here.

As a side note do you have Kinova arm and hand?
The following code is used to talk to my j2s6s300 arm and hand.
Unless you have another suggestion.
https://github.com/mpatalberta/drake-jaco-driver

Thanks,

Patrick Brown

mpatalberta, 2021-01-25:
Ok so I have been able to create and a new j2s6s300_sphere_collison.urdf,
The drake visualization look correct.
The jaco simulator does not appear to work for the simulation mode.

Can some one suggest what need be changed in jaco_controller ,jaco_simulator and jaco_move.
I have checked all my code into my pull

mpatalberta, 2021-01-25:
Ok so I have been able to create and a new j2s6s300_sphere_collison.urdf,
The drake visualization look correct.
The jaco simulator does not appear to work for the simulation mode.

Can some one suggest what need be changed in jaco_controller ,jaco_simulator and jaco_move.
I have checked all my code into my pull
I also noticed in jaco_simulator the that 7 links are written to instead of the 6 for my arm geomtry.

mpatalberta, 2021-01-25:
I had found that jaco_controller,jaco_simulator move had to be changed to read the j2s6s300_sphere_collision.urdf where changed as default.

The jaco_constants.h needed to be changed to 6 joint from 7. This  code need to get the joint number programmatically and not use a hardcoded value

jwnimmer-tri, 2021-02-04:
As far as I can understand the above discussions, this is neither a bug report nor feature request for Drake -- it is rather a discussion about how the get various tools (Drake & otherwise) to work, and how they should be used.  As such, it should be a StackOverflow post, not a GitHub issue.  I'm closing it for that reason.

closed 2021-02-04 by jwnimmer-tri