RobotLocomotion/drake #14559: I have ch

mpatalberta, 2021-01-22:
I have checked in my code changes and changes to the urdf files that are used
The jaco_controller gives the incorrect number of joints should be 6 and not 7

nl@tnl:~/drakeother/drake$ bazel-bin/examples/kinova_jaco_arm/jaco_controller
[2021-01-22 12:21:11.444] [console] [warning] Currently MultibodyPlant does not handle joint limits for continuous models. However some joints do specify limits. Consider setting a non-zero time step in the MultibodyPlant constructor; this will put MultibodyPlant in discrete-time mode, which does support joint limits.
[2021-01-22 12:21:11.444] [console] [warning] Joints that specify limits are: `j2s6s300_joint_1`, `j2s6s300_joint_2`, `j2s6s300_joint_3`, `j2s6s300_joint_4`, `j2s6s300_joint_5`, `j2s6s300_joint_6`, `j2s6s300_joint_finger_1`, `j2s6s300_joint_finger_tip_1`, `j2s6s300_joint_finger_2`, `j2s6s300_joint_finger_tip_2`, `j2s6s300_joint_finger_3`, `j2s6s300_joint_finger_tip_3`.
num_joints:7
num_fingers:3
plan_source->plant().num_positions()12
plan_source->plant().num_velocities()12
abort: Failure at examples/kinova_jaco_arm/jaco_controller.cc:77 in DoMain(): condition 'plan_source->plant().num_positions() == num_joints + num_fingers' failed.

How do you build the urdf files from the kinova arm j2s6s300_standalone.xacro file.
I tried running ros xacro to urdf but that did not work.
I get an error:
[2021-01-21 14:04:19.196] [console] [warning] Currently MultibodyPlant does not handle joint limits for continuous models. However some joints do specify limits. Consider setting a non-zero time step in the MultibodyPlant constructor; this will put MultibodyPlant in discrete-time mode, which does support joint limits.
[2021-01-21 14:04:19.196] [console] [warning] Joints that specify limits are: `j2s6s300_joint_1`, `j2s6s300_joint_2`, `j2s6s300_joint_3`, `j2s6s300_joint_4`, `j2s6s300_joint_5`, `j2s6s300_joint_6`, `j2s6s300_joint_finger_1`, `j2s6s300_joint_finger_tip_1`, `j2s6s300_joint_finger_2`, `j2s6s300_joint_finger_tip_2`, `j2s6s300_joint_finger_3`, `j2s6s300_joint_finger_tip_3`.
terminate called after throwing an instance of 'std::runtime_error'
  what():  This multibody tree already has a mobilizer connecting these two bodies. More than one mobilizer between two bodies is not allowed
Aborted (core dumped)

SeanCurtis-TRI, 2021-01-25:
@mpatalberta   Several things:

1. We need you to clean up the issue title. As is, the complaint isn't clear.
2. This feels a bit like a stack-overflow question more than a Drake issue (although that sense may change based on clarification of the issue text).
3. "I have checked in my code changes and changes to the urdf files that are used".  It's not clear what this means. Do you have a unique local version of code and urdf that is unavailable to other Drake developers? That could make troubleshooting this difficult.
4. Are you aware that `examples/kinova_jaco_arm/jaco_controller` has optional command line arguments: `--num_joints` and `--num_fingers`? if you've modified the urdf to reduce the number of joints from the default number, you can override it at the invocation.
5. @amcastro-tri It seems there is a bug in the reported error message: " This multibody tree already has a mobilizer connecting these two bodies. More than one mobilizer between two bodies is not allowed". It would be better if "these two bodies" were better communicated.

SeanCurtis-TRI, 2021-01-25:
Assigning to @sherm1 for subsequent disposition.

SeanCurtis-TRI, 2021-01-25:
Ah...this appears to be spill over from #14556. I'm closing this as duplicate of that. Any information here that is relevant there, should be placed there.

I'd take a quick opportunity to emphasize (particularly given #14556) that this really is more of a stackoverflow kind of topic more than a drake *issue*.

closed 2021-01-25 by SeanCurtis-TRI

rpoyner-tri, 2021-01-25:
@SeanCurtis-TRI FWIW, my https://github.com/RobotLocomotion/drake/pull/14534 will likely give better error messages in cases like this. I suspect the branch that @mpatalberta is using is a bit older than when that merged.