RobotLocomotion/drake #14561: Numerical instability for PiecewiseQuaternionSlerp

rcywongaa, 2021-01-24:
Consider the following program
```
from pydrake.all import PiecewiseQuaternionSlerp, Quaternion
import numpy as np

q1 = np.array([1.0, 0.0, 0.0, 0.0])
q2 = np.array([-0.5, -0.5, 0.5, 0.5])
slerp1 = PiecewiseQuaternionSlerp(breaks=[0, 1], quaternions=[Quaternion(q1), Quaternion(q2)])
for i in np.linspace(0, 1, 11):
    print(f"1: {Quaternion(slerp1.value(i)).wxyz()}")

q3 = np.array([1.0, 0.0, 0.0, 0.0])
q4 = np.array([-0.5, -0.5, 0.4999999999999999, 0.5000000000000001])
slerp2 = PiecewiseQuaternionSlerp(breaks=[0, 1], quaternions=[Quaternion(q3[0:4]), Quaternion(q4[0:4])])
for i in np.linspace(0, 1, 11):
    print(f"2: {Quaternion(slerp2.value(i)).wxyz()}")
```
Output:
```
1: [1. 0. 0. 0.]
1: [ 0.9945219   0.06034954 -0.06034954 -0.06034954]
1: [ 0.9781476   0.12003787 -0.12003787 -0.12003787]
1: [ 0.95105652  0.17841104 -0.17841104 -0.17841104]
1: [ 0.91354546  0.23482951 -0.23482951 -0.23482951]
1: [ 0.8660254   0.28867513 -0.28867513 -0.28867513]
1: [ 0.80901699  0.33935797 -0.33935797 -0.33935797]
1: [ 0.74314483  0.38632274 -0.38632274 -0.38632274]
1: [ 0.66913061  0.42905487 -0.42905487 -0.42905487]
1: [ 0.58778525  0.46708618 -0.46708618 -0.46708618]
1: [ 0.5  0.5 -0.5 -0.5]
2: [1. 0. 0. 0.]
2: [ 0.9945219   0.06034954 -0.06034954 -0.06034954]
2: [ 0.9781476   0.12003787 -0.12003787 -0.12003787]
2: [ 0.95105652  0.17841104 -0.17841104 -0.17841104]
2: [ 0.91354546  0.23482951 -0.23482951 -0.23482951]
2: [ 0.8660254   0.28867513 -0.28867513 -0.28867513]
2: [ 0.80901699  0.33935797 -0.33935797 -0.33935797]
2: [ 0.74314483  0.38632274 -0.38632274 -0.38632274]
2: [ 0.66913061  0.42905487 -0.42905487 -0.42905487]
2: [ 0.58778525  0.46708618 -0.46708618 -0.46708618]
2: [-0.5 -0.5  0.5  0.5]
```

While I'm aware that `[0.5, 0.5, -0.5, -0.5]` and `[-0.5, -0.5, 0.5, 0.5]` represent the same rotation, it seems unintuitive that the slerp will return quaternions with a large jump in element values.

This was tested with the latest v0.26.0 release

SeanCurtis-TRI, 2021-01-25:
@RussTedrake As the primary author of this code (at least as far as the existing history knows), I'm assigning this to you. Feel free to pass it along as appropriate.

RussTedrake, 2021-01-25:
Actually, I only touched the API... @siyuanfeng-tri is the real author.  @siyuanfeng-tri , want to take a look?

siyuanfeng-tri, 2021-01-26:
sure.

siyuanfeng-tri, 2021-01-26:
Thanks for reporting this.
The issue is coming from you getting the interpolated Quaternion as a matrix (through `slerp2.value(t)`) then converting it to quaternion with Eigen's constructor. Because we didn't expose the full c++ API to python for getters like this https://github.com/RobotLocomotion/drake/blob/master/common/trajectories/piecewise_quaternion.h#L91, which doesn't have this issue.. I will need to submit a patch to expose those.

FYI, drake's matrix to quaternion conversion does force a sign choice for quaternions, https://github.com/RobotLocomotion/drake/blob/master/math/rotation_matrix.h#L587. You could use that if that's appropriate.

siyuanfeng-tri, 2021-01-28:
should be patched.

closed 2021-01-28 by siyuanfeng-tri