RobotLocomotion/drake #14566: value: Why does TypeHash<> hate nested types in a templated class?

EricCousineau-TRI, 2021-01-26:
For more context, see @hongkai-dai's question here:
https://drakedevelopers.slack.com/archives/C3YB3EV5W/p1611620611000900?thread_ts=1611620465.000800&cid=C3YB3EV5W

Basically, `Value<TemplatedType<S>::Inner>` doesn't play well with `TypeHash<T>`.

Min-repro: ebea59f4d3
https://github.com/RobotLocomotion/drake/compare/master...EricCousineau-TRI:issue-value-inner-templated-boo?expand=1

@jwnimmer-tri I've assigned ya for now; feel free to reassign?

jwnimmer-tri, 2021-01-26:
I'm happy to have it assigned to me, but I'm unclear on what the call to action is?
- Have Drake CI fail-fast if first-party Drake code contains such a defect?
- Improve the error message so that a developer can more quickly understand and repair the defect?
- Improve TypeHash so that it can handle this case without complaint?

EricCousineau-TRI, 2021-01-26:
Ah, sorry! The call to action is ideally "Improve TypeHash so that it can handle this case without complaint"
Barring that, ideally we have this failure case denoted in a unittest, with advice on how to mitigate this (e.g. don't use the nested type of a template class, etc.)

Does that work?

jwnimmer-tri, 2021-01-26:
Fixing it for real (option 3) requires writing some approximation of an actual typename parser that supports nesting, which used to be somewhat awkward in the `constexpr` language subset in '14.  Maybe with '17 or '20 now it's easier, but I'd argue it's still low bang for the buck.  In 80%+ of the time this error message appears, the user's code code is already defective (nesting a scalar-agnostic helper struct within a scalar-specific outer class, as was the case in the OP).

Thus, I'll take it as just option (2) above -- improve the diagnostic message, probably pointing to a nice example.

PR #14709 (2021-03-03): common: Improve AbstractValue error message suggestions
Closes #14566.

---

Here is an example of the new report, taken from the output of ValueTest.NonTypeTemplateParameter:

[console] [warning] The drake::test::(anonymous)::UnadornedAnonEnumTemplate<(drake::test::(anonymous)::AnonEnum)0> class is incompatible with the typename hasher that provides the type-erasure checking for AbstractValue casts, most likely because the problematic class mixes template parameters with nested classes or non-type template parameters. As a result, operations on Value<drake::test::(anonymous)::UnadornedAnonEnu [... cut]
closed 2021-03-03 by soonho-tri