RobotLocomotion/drake #14568: drake visualizer shows no activity when you run the defaultg j2s7s300_sphere_collision.urdf

mpatalberta, 2021-01-26:
Of the current branch when i run the examples/kinova_jaco_arm  and follow the example in the readme no movement of the 
arm is shown to visualizer

jwnimmer-tri, 2021-01-26:
When I run these commands, I do see motion:
```console
bazel build //tools:drake_visualizer //examples/kinova_jaco_arm/...
bazel-bin/tools/drake_visualizer &
bazel-bin/examples/kinova_jaco_arm/jaco_simulation &
bazel-bin/examples/kinova_jaco_arm/jaco_controller &
bazel-bin/examples/kinova_jaco_arm/move_jaco_ee &
```

Can you confirm those are exactly what you're trying?

mpatalberta, 2021-01-26:
Mr. Nimmer,
That is the order.
The robot arm  appears in the drake visualizer but never moves when the move_jaco_ee is done.
Thanks,
Patrick

[External Email]

- CONFIDENTIAL-

This email and any files transmitted with it are confidential, and may also be legally privileged. If you are not the intended recipient, you may not review, use, copy, or distribute this message. If you receive this email in error, please notify the sender immediately by reply email and then delete this email.

mpatalberta, 2021-01-26:
Mr. Nimmer,
I found my problem.
I was getting closer to the and I can see the rotation and movement.

Pat,

mpatalberta, 2021-01-26:
Found the problem. Expand the view (get closer) to the arm and you wil see the movement.

closed 2021-01-26 by mpatalberta

mpatalberta, 2021-01-28:
Mr. Nimmer
Found my problem.
Thanks,
Patrick