RobotLocomotion/drake #14570: pydrake sphinx: Bad word wrapping with Doxygen -> Sphinx bulleted lists

EricCousineau-TRI, 2021-01-26:
As of `nightly-release` @ 2f41e7f8b:
https://drake.mit.edu/pydrake/pydrake.systems.sensors.html#pydrake.systems.sensors.CameraInfo
> [image]

https://drake.mit.edu/pydrake/pydrake.geometry.render.html#pydrake.geometry.render.ClippingRange
> [image]

jamiesnape, 2021-01-26:
This has always been the case, unfortunately. Sphinx expects one exception per line, and we never got around to unwrapping lines.

EricCousineau-TRI, 2021-01-26:
Yup, didn't mean to imply this is a novel failure. May add weight to considering other solutions for web deployment, like Breathe.

mwoehlke-kitware, 2022-03-01:
Can someone briefly explain how these are generated? (As in, hints to someone trying to fix this where to go looking?)

jwnimmer-tri, 2022-03-01:
First to set the stage, this ticket is not very important in the overall landscape of Drake.  Per a chat with @BetsyMcPhail it only got pulled onto the project board as an onboarding ticket for @alesgenova.  It's fine to have @alesgenova (or any other new onboarding Kitware contributor) work on it, but I'd prefer if we could have you (@mwoehlke-kitware) focus the more important tickets from the board.

To the question, though:

In the https://github.com/RobotLocomotion/drake/tree/master/tools/workspace/pybind11 folder, the files `mkdoc.py` and `mkdoc_comment.py` are most of what controls the python docstring formatting, when converting C++ comments into Napoleon syntax.

The https://drake.mit.edu/doxygen_cxx/group__python__bindings.html page also has a bit more detail about the documentation generation.

jwnimmer-tri, 2023-10-05:
Closing this as not planned.  The current documentation transmogrification system is fundamentally broken and the only salvation will be to re-implement it from scratch using actual specifications and proper parsing.

closed 2023-10-05 by jwnimmer-tri