RobotLocomotion/drake #14575: doc: Should re-do Jekyll styling for reworked Drake website, handle licensing

EricCousineau-TRI, 2021-01-26:
As part of #13832, should rework Jekyll styling into a source-controlled form that has no license ambiguity and enables due-diligence agency of modifying the website (e.g. will assume devs know how to look at W3C for HTML + CSS stuff, Jekyll docs).

Artifacts to reconcile:

- Preview website: https://drake.sky.a2hosted.com/
- Zip archive on [Drake Slack, `#website` channel](https://drakedevelopers.slack.com/archives/C01879RNKQW/p1603914850001300) with artifacts that need to be reconstructed
- Converted reST to Jekyll Markdown (either #14479 or #14488 are good starting points for sample content)

Steps:
- Use a Jekyll "Getting Started" tutorial to generate sample website
- Establish baseline website Jekyll setup
- Incrementally replay changes, ensuring that we have licensing

Per meeting, @IanTheEngineer and I will take this. I will spearhead, and then ask Ian for help / review.

FYI @jwnimmer-tri @ToffeeAlbina-TRI @ggould-tri @rpoyner-tri

EricCousineau-TRI, 2021-01-26:
@BetsyMcPhail Can you advise on which PR we should use for sample content to target styling against?

EricCousineau-TRI, 2021-01-26:
Internet's going slow; going to timebox to 30min, trying out re-styling with #14488

jwnimmer-tri, 2021-01-26:
> Can you advise on which PR we should use for sample content to target styling against?

To obtain some sample markdown files (i.e., some lorem ipsum) so as to have actual text to display, in #14488 you only have ~one file, `getting_help.md`.

If you want _many_ markdown files at once, the draft in #14479 has a whole bunch, and is probably a better starting place to snag a few.  (You probably don't need them all.)

jamiesnape, 2021-01-27:
@BetsyMcPhail has the licenses are separated out, and everything fixed, but that may only exist locally now. You can reach out to her and she can push the latest version somewhere that is laid on a platter for you all to review,

EricCousineau-TRI, 2021-01-27:
Ooh, yes, that would be awesome!
Betsy, would be able to push what you have in terms of licensing?

jamiesnape, 2021-01-27:
She has literally everything, so all you are going to need to do is cherry-pick and review. The markdown she has should be an up-to-date sync with the rst modulo maybe a week. Should appear shortly.

BetsyMcPhail, 2021-01-27:
The branch with all of my changes is available [here ](https://github.com/BetsyMcPhail/drake/tree/jekyll-website).

EricCousineau-TRI, 2021-02-01:
Thanks! Just to check, can you tell me how you generated the base support files in f7866d7cdd?

The commit message states that some of them come from the web devs that made the redux, but it's unclear. Also, for licensing, it's unclear if this came from a template setup from Jekyll; is it possible to clarify that?

jamiesnape, 2021-02-01:
> Also, for licensing, it's unclear if this came from a template setup from Jekyll; is it possible to clarify that?

Unlikely, but given the license of a tool, it does not imply anything about the license of the output, regardless.

jamiesnape, 2021-02-01:
BTW whether the web developers dropped a new file into a directory or edited such file or whether Betsy did, they are both work for hire for the same client, so the copyright and license of new files would be identical.

jamiesnape, 2021-02-01:
(unless the contract with the web developers specified otherwise, which would be very unusual)

jwnimmer-tri, 2021-02-01:
Yes, it was all work for hire, so you're gist is correct -- TRI owns the copyright.  The only question is whether anyone copied files from _elsewhere_ and dropped them into the folder.  Anything created out of thin air is fine, anything downloaded from the internet needs a confirmation of our rights.

jamiesnape, 2021-02-01:
I think that is documented in any case that it could be reasonably inferred. Betsy will confirm.

BetsyMcPhail, 2021-02-01:
Anything in the developers' work that was clearly third party code has either been removed or is part of its own commit with appropriate license and documented source. 

In commit `f7866d7`:
* BUILD.bzl --> modified existing drake file
* _config.yml, css/style.css --> drake-specific changes + developers' code
* _layouts/default.html --> removed unsupported content (e.g. fonts, facebook  & twitter)
* Everything else--> copied directly from developers work with exception of whitespace changes (e.g. tabs to spaces). There is nothing in here that looks like it was copied from a third-party source.

EricCousineau-TRI, 2021-02-01:
> Unlikely, but given the license of a tool, it does not imply anything about the license of the output, regardless.

True, but I would like to confirm (e.g. if the template generation produces a derivative of licensed template files, would be nice to know?)

jamiesnape, 2021-02-01:
You will need to ask the web developers, then. It is a lot of work for no consequence.

EricCousineau-TRI, 2021-02-01:
Trust but verify is the motto that I will go by. Verify will take place in a PR that I start, but leveraging Betsy's work, where we will go through line-by-line to identify any possible licensing issues.
The PRs will most likely break up f7866d7 into slightly smaller chunks.

I will ask web devs if they remember, but it won't be on critical path.

EricCousineau-TRI, 2021-02-02:
Identified base template from `jekyll new` for v3.1.6: https://github.com/jekyll/jekyll/tree/v3.1.6/lib/site_template

This does not look anything like the base that is used for our newer website.

EricCousineau-TRI, 2021-02-03:
(As noted in integration PR)
Per Slack conversation: https://drakedevelopers.slack.com/archives/C01879RNKQW/p1612374479018500?thread_ts=1612374025.015800&cid=C01879RNKQW
The website template came from: https://jekyllthemes.io/theme/made-portfolio-jekyll-theme

We will now take steps in doing a best-effort purge of this template from our Git history.

EricCousineau-TRI, 2021-02-17:
Working: Adding video here to avoid polluting git history:
https://user-images.githubusercontent.com/26719449/108152577-4d14f300-70a7-11eb-88c5-5da0c39b1e24.mp4

PR #14657 (2021-02-18): doc: Incorporate new Jekyll website style
Resolves #14575

EchoCharlie did another revision that was simplified to minimal CSS / JavaScript, starting from a blank Jekyll template.
I then took it, simplified the setup (I inlined `_homeblocks`, `_developers`, etc.), and did a more thorough job for attribution. I also ported some Sphinx styles over (e.g. warnings / notes), and other minutae (if you're curious, I can add you to the repo).

Requires:

- [x] #14613 Novel images
- [x] #14614 Third-party images
- [x] #14658 Markdown shuffle
- [x] #14659 Minor Markdown improvement
- [x] #14662 Add'l th [... cut]
closed 2021-02-18 by EricCousineau-TRI