RobotLocomotion/drake #14577: Transmogrify rst documentation files to md

jwnimmer-tri, 2021-01-26:
For now, just grabbing an issue number so that I can cite this as a subtask from #13832.  I'll rewrite this description with more specifics shortly.

#14479 has an incomplete draft of this work (+ other style changes).

ggould-tri, 2021-01-27:
This sounds right to me; our objective is to make an initial PR that consists only of formatting changes and which appears as much as possible perfectly mechanical.

FYI @SeanCurtis-TRI

ggould-tri, 2021-01-27:
(Also just to be clear, the resulting diff might be sane in reviewable as a copy revision and a diff revision, or it might require offline review with semantic diff tools; in the (more likely) latter case we will select reviewers who are comfortable with such tools)

jamiesnape, 2021-01-27:
We have the markdown converted to rst already, but if you want to reinvent the wheel that is fine.

ggould-tri, 2021-01-27:
I don't expect to reinvent the wheel; I hope that this issue in practice is just a matter of structuring what you already have in easily reviewable form.

jamiesnape, 2021-01-27:
I think Betsy even minimized the diff so each page, so I will ping you when she is able to push the latest complete branch, which may have disappeared in the back and forth of the various PR's.

jamiesnape, 2021-01-27:
Probably the release notes for v26.0 onwards is all you really need to do yourself.

BetsyMcPhail, 2021-01-27:
The branch with all of my changes is available [here](https://github.com/BetsyMcPhail/drake/tree/jekyll-website) . 

The branch was created around January 12th, so there will be some updates that need to be made.

For most files, a diff between the md and rst should be straightforward.

Of note:
* In the source provided by the original website contractors, the ``developers`` page was split into multiple pages in order to create a table of contents. This looked nice but was not maintainable. In the ``jekyll-website`` branch, developers is back to a single page with no table of contents.
* Similarly, the index page has been split into multiple documents within the ``_homeblocks`` directory. This is in a separate commit.

EricCousineau-TRI, 2021-02-16:
Additional action item:
- Synchronize #14646

ggould-tri, 2021-02-16:
TODO items remaining, actionable now:
 * [ ] update md for #14646
 * [ ] Rescue `_homeblocks` directory from @BetsyMcPhail 's branch (link above is broken; need a new branch link)
 
TODO after jekyll rendering is working:
 * [ ] Fix any broken jekyll headmatter on md files.

jamiesnape, 2021-02-16:
You should be able to create a single markdown file in `_homeblocks` with just
```
---
title: Random Title
---

Random content.
```
and I think it will at least unblock the build.

jamiesnape, 2021-02-16:
Actually, the original files were just chunks of the previous main page. I can create them for you if @EricCousineau-TRI does not have a copy.

jwnimmer-tri, 2021-02-16:
I think #14479 still has a copy.  The personal branch was deleted, but the pull request still has all the content.  There's also (in the `#website` slack channel) the original zipfile from the contractors to use for reference.

jamiesnape, 2021-02-16:
I do not think @BetsyMcPhail changed anything, but even if she did, the originals should still work.

BetsyMcPhail, 2021-02-16:
I didn't change anything of substance in the ``_homeblocks`` directory.

EricCousineau-TRI, 2021-02-16:
I can do a skeleton `_homeblocks` PR.

Not loving what the file organization looks like, but may give up shortly.

EricCousineau-TRI, 2021-02-16:
Actually, scratch all that. I'd suggest waiting until I have the PR ready.

(I just have two more items left - simplify TOC-style logic, and minimize directory tree differences, 'cause the current setup is still a tad confusing.)

jwnimmer-tri, 2021-02-23:
@ggould-tri Is this issue ready to close?

It looks like _"Fix any broken jekyll headmatter on md files."_ is the only item not checked off.

ggould-tri, 2021-02-23:
Yes, I believe that with #14687 we are where we need to be:  Future edits will have a reasonably working rendered world in hand that is up-to-date with the markdown world; transition to Jekyll is now primarily aesthetic rather than content work.

closed 2021-02-23 by ggould-tri