RobotLocomotion/drake #14578: Bug with hydroelastic visualization

amcastro-tri, 2021-01-26:
For details refer to the internal TRI [Slack conversation here](https://tri-internal.slack.com/archives/C2BMF3K28/p1611676244032300).

We observed that in hydroelastic simulations leading to multiple contact patches, one of the patches seems to include all other patches, therefore repeating the information. More precisely, in all tests it seems it is the last patch in the `Hydroelastic Contact` panel in Drake viz that seems to show this problem.
We verified that the data in `ContactResults` (produced by `MBP::CalcContactResultsContinuousHydroelastic()`) and in `lcmt_contact_results_for_viz` (produced by `ContactResultsToLcmSystem::CalcLcmContactOutput()`) is correct and does not show this duplication (see [commit used to verify this here](https://github.com/amcastro-tri/drake/commit/ce5466ddc81f263c044ea803869c0a23c0f7a149)).

Therefore we believe this issue is purely related to visualization and does not affect the dynamics of the system.

SeanCurtis-TRI, 2021-01-26:
TODO: Create scenario to reproduce the issue (e.g., create a unit test. :D)

xuchenhan-tri, 2021-04-30:
The root cause of the issue seems to be the DebugData variable being defined outside of the loop for contact surfaces. See [here](https://github.com/RobotLocomotion/drake/blob/9a1b66f351b470bad0a6f4827eb8069dfab48dda/tools/workspace/drake_visualizer/_drake_visualizer_builtin_scripts/show_hydroelastic_contact.py#L635) for the source.

There has been a few changes to hydroelastic visualization since this issue was filed. In particular, #14622 alleviates the issue to some degree by passing the pressure field data directly to VTK instead of through DebugData. So the issue described can't seem to be completely reproduced. The contact forces and moments, however, are still being visualized through DebugData and the bug still persists. 

The fix should be easy (moving the definition of the DebugData to inside the loop). I'm working on stripping down my reproducing case to a minimal unit test and I'll make a PR for a fix soon.

amcastro-tri, 2021-04-30:
nice! thanks @xuchenhan-tri!

PR #15000 (2021-05-10): [visualization] Fix bug in hydroelastic visualization
Fix #14578. Make sure that contact spatial force is no longer duplicated 
in hydroelastic visualization. Modify the simple_contact_surface_vis 
example so that it verifies the fix.

Make two other incremental improvements to hydroelastic visualization:
1. Instead of visualizing the contact surfaces as two offset surfaces,
visualize the actual contact surface as a single entry in scene browser
2. Give a more descriptive name to the spatial force entry in the scene
browser.

---
This change is [[image]](https://reviewable.io/reviews/robotloco [... cut]
closed 2021-05-10 by xuchenhan-tri