RobotLocomotion/drake #14584: Improve efficiency of calculations involving inverses of rotation matrix and rigid transform.

mitiguy, 2021-01-27:
RotationMatrix and RigidTransform are core Drake classes and can be expected to be found in inner loops of calculations.  To  speed calculations by eliminating a transpose (and possibly a copy), implement efficient Rotational Matrix (and RigidTransform) methods.

Implementation details are important.  The method could have a name, an operator, or we could introduce an intermediate type (e.g., RotationMatrixInverse) that, when pre or post multiplied, behaves efficiently.

sherm1, 2021-01-27:
FYI I posted an attempt at a fast RotationMatrix composition (3x3 matrix multiply) method using AVX instructions in a Slack thread [here](https://tri-internal.slack.com/archives/C2BMF3K28/p1611728450054200?thread_ts=1611619991.017900&cid=C2BMF3K28). Seems to run about twice as fast as fully-optimized straight C++.

DamrongGuoy, 2021-01-27:
Thank you for giving attention to this issue.  An example that shows up in my profiling task is at the last line (`X_GA.inverse() * X_GH * X_HB`) of this code snippet:

drake/geometry/proximity/obb.cc
```C++
bool Obb::HasOverlap(const Obb& a, const Obb& b,
                     const math::RigidTransformd& X_GH) {
  // The canonical frame A of box `a` is posed in the hierarchy frame G, and
  // the canonical frame B of box `b` is posed in the hierarchy frame H.
  const RigidTransformd& X_GA = a.pose();
  const RigidTransformd& X_HB = b.pose();
  const RigidTransformd X_AB = X_GA.inverse() * X_GH * X_HB;
...
```
https://github.com/RobotLocomotion/drake/blob/54f23a2e2a7f5e840683c5341b562712ae21127b/geometry/proximity/obb.cc#L25

This function was called 140,994,468 times in a 20-second simulated time. It checks whether there is an overlap between two oriented bounding boxes of some elements of two meshes.

jwnimmer-tri, 2021-11-12:
We have a function `RotationMatrix::InvertAndCompose` and a function `RigidTransform::InvertAndCompose` now.  What additional work remains before we can close this issue?

sherm1, 2021-11-12:
I think PR #15119 (and its predecessors) handled most of what the OP wanted to achieve, though maybe not all. If we discover further hot spots involving these classes we can attempt to go further. For now we should consider this closed.

closed 2021-11-12 by sherm1