RobotLocomotion/drake #14590: py mbp: Getting hard exit from "Failure at external/drake/multibody/plant/tamsi_solver.cc:217 in SolveQuadraticForTheSmallestPositiveRoot(): condition 'Delta > 0' failed."?

EricCousineau-TRI, 2021-01-28:
In some Python code in Anzu, I'm running MBP with some crazy neural network whatever visuomotor stuff, and at some point I get the following *hard* error, using 6ee5e9325821277a62bd5cd5456ccf02ca25dab7
```
Failure at external/drake/multibody/plant/tamsi_solver.cc:217 in SolveQuadraticForTheSmallestPositiveRoot(): condition 'Delta > 0' failed.
```

This should at least be an `Exception` in Python-land? (e.g. #5268).
I'm not sure if I can make a min-repro, but I'll at least see if I can convert this to a softer exception.

May re-assign if there's more nuance here.

FYI @amcastro-tri @sherm1

EricCousineau-TRI, 2021-01-28:
I should double-check that it's not just a `SystemExit` condition as well.

amcastro-tri, 2021-01-28:
Those are rare. But usually are related to the solver already in the process of diverging. It does happen if the condition you are trying to solver for takes the solver to unstable land. Might also be a new bug, but that's the usual case.

sherm1, 2021-01-28:
Is there a way to catch that condition @amcastro-tri to produce a better error message?

mpetersen94, 2021-01-28:
Also experiencing this bug but I'm using a peg from September so it isn't newly introduced.  Would not be surprised if it is the result of unstable sim.

EricCousineau-TRI, 2021-01-28:
Yeah, I think integrator explosions / instability probably wouldn't be surprising (this is testing a policy that comes from training on top of randomly initialized weights); the main crux of this issue is namely the user-unfriendly mode of killing the interpreter.

sammy-tri, 2021-01-29:
I've been hitting these occasionally too, usually when making model changes.  It would be helpful if there were a debugging strategy (and/or a more helpful message).  My current strategy is to give up and say "well, I guess I don't get to try that."

amcastro-tri, 2021-02-01:
What I could do is to make `TamsiSolver::SolveWithGuess()` detect this condition and return something like `TamsiSolverResult::kDivergence`. This would then lead to printing the much nicer [message here](https://github.com/RobotLocomotion/drake/blob/ce36a0fba54442d425c8a800638e71355f2e68e4/multibody/plant/multibody_plant.cc#L2042) explaining not only what happened but also providing possible solutions.

Would that be an acceptable solution?

EricCousineau-TRI, 2021-02-01:
That would be awesome!!!

jwnimmer-tri, 2023-01-05:
> I should double-check that it's not just a SystemExit condition as well.

It was.  I've fixed that in #18543.

RussTedrake, 2023-05-14:
Based on the experience in class projects last fall (https://github.com/RobotLocomotion/drake/issues/18452) and again this spring, I believe that this Delta>0 error is not rare at all -- my impression is that it occurs more often than the nicer error message that @amcastro-tri eluded to above. I think this is a major usability issue, and would recommend that we raise it's priority to high (and likely that we re-assign the issue and resolve it quickly).

@rpoyner-tri, @jwnimmer-tri, wdyt?

jwnimmer-tri, 2023-05-14:
To make sure we're on the same page -- by "resolve it" do you mean "throw an error that says something more specifically about solver divergence"?  Specifically it's *not* a goal of this ticket to stop throwing, but rather just throw something that isn't confusing?

Or in other words, implementing @amcastro-tri's suggested fix.

jwnimmer-tri, 2023-05-14:
See #19407 for the proposed fix (as of yet untested). _None_ of the bug reporting upthread provided a reproducer, so I'll try to invent a test case and pray it fixes the user problems.

jwnimmer-tri, 2023-05-14:
Okay, I pushed a silly unit test.  If anyone has a sample dynamics input that could be used as an end to end test, please share.  If not, we can add that in later once we have one.

RussTedrake, 2023-05-14:
There are some maybe(?) candidates on SO:
https://stackoverflow.com/search?q=%5Bdrake%5D+SolveQuadraticForTheSmallestPositiveRoot

It seems that both of these have to do with zero inertias in the internal of the mechanism...

jwnimmer-tri, 2023-05-15:
Thanks, I hadn't checked there yet.  However, neither seems to have a reproducible example.  For now, I think we'll just hope that the small unit test in the PR is good enough to meet the need.

PR #19407 (2023-05-16): [multibody] Improve error message for !(Delta > 0)
Closes #14590.
Closes #19416.

I suspect someone is feeding a NaN into this arithmetic and it just happens to get all the way to the `sqrt` before we notice.  It's likely that diagnosing the NaN earlier would be better, but anyway this at least somewhat improves the message in this case.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19407)