RobotLocomotion/drake #14593: when running a real kinova arm the jaco_controller semd random data

mpatalberta, 2021-01-28:
externall driver for real kinova arm from https://github.com/mpatalberta/drake-jaco-driver
/example/kinova_jaco/jaco_controller.cc

The jaco_controller.cc initially received correctly the current angles of the actuators in the and broadcasts that information
Then the jaco_controller.cc starts outputing some very small number for about 10 - 20 second.
The sudden;y a series of non-zero numbers are send as angular velocity values which causes the arm to in a in correct manner.

Patrick Brown

sammy-tri, 2021-01-28:
Reproduced locally using a 7DOF Jaco arm.  The issue does not appear to occur when running `jaco_controller` against a simulation.  

Also, the controller was sending joint velocities over the LCM command channel without a corresponding change in positions, which is a bit odd.  Will investigate further.

mpatalberta, 2021-01-28:
Mr. Creasey,

I have a real j26s300 kinova arm.

A couple of ques Reproduced locally using a 7DOF Jaco arm. The issue does not appear to occur when running jaco_controller against a simulation.

I do not see the problem in mine either using jac_simulator

Also, the controller was sending joint velocities over the LCM command channel without a corresponding change in positions, which is a bit odd. Will investigate further

Is their a message pump where pushing out LCM command without any movement somewhere?

I have few questios:
1 do I need jaco_simulator if I already have a drake based kinova_driver [Sammy Try drake_jaco_driver].
                You can access the code at github  mpatalberta/kinova_jaco_driver

2 Is correct that if I want to  communicate to drake visualizer this code exists somewhere other than in jaco_simulator

3 When I run either the j2s7s300 or j2s6s300 both work correct when the jaco_simulator used and not the drake_jaco_driver from
Thanks,
Patrick

[External Email]

- CONFIDENTIAL-

This email and any files transmitted with it are confidential, and may also be legally privileged. If you are not the intended recipient, you may not review, use, copy, or distribute this message. If you receive this email in error, please notify the sender immediately by reply email and then delete this email.

sammy-tri, 2021-01-28:
I think I'm pretty close to having a patch which will fix the issues with `jaco_controller`.  The gains on the PD controller were too high (I don't think we want a D term at all, that was causing the unexpected velocity commands due to (I think)) noise in the reported velocities form the Jaco), and that instability was being fueled by the controller publishing an excessive number of command messages (I'm not exactly sure why that's a problem, but fixing it made matters better).  I'll tag you in the PR once I have something ready.

> do I need jaco_simulator if I already have a drake based kinova_driver

I test both in simulation and on hardware because I find both convenient for me, but, no, you don't need the simulator if you just want to run the hardware.

> Is correct that if I want to communicate to drake visualizer this code exists somewhere other than in jaco_simulator

Correct.  The simulator uses the same visualization facilities provided by `MultibodyPlant` + `SceneGraph` that other drake systems use.

mpatalberta, 2021-01-29:
Mr. Creasey,

Maybe you can help  what are the different plan and trajectory planners are supported.

I am more familiar with ROS where you can explicitly see CHOMP ,STOMP and so on.

On drake I do not see these type path planners that are supported. What path planners are supported.

Patrick Brown
Zebra technologies

[External Email]

mpatalberta, 2021-01-29:
Mr. Creasey,
Looking at the Kinova-ros code a file exists kinova-ros<https://github.com/Kinovarobotics/kinova-ros>/kinova_control<https://github.com/Kinovarobotics/kinova-ros/tree/master/kinova_control>/config<https://github.com/Kinovarobotics/kinova-ros/tree/master/kinova_control/config>/j2s6s300_control.yaml
Make sure you look at the master
https://github.com/Kinovarobotics/kinova-ros

Maybe you can take the pid values from the supplied yaml file for the relevant arm.
Pat,

mpatalberta, 2021-02-02:
I have gone and modify the pid values in the kinova_controller. The arm appears initially stable  until you send a move_jaco.
When this is done nothing happens initially but suddenly it moves.

I have changed the move_jaco location to be where the robot would be for home.
The x,y,z ordinates are correct however the euler angles are wrong.

sammy-tri, 2021-02-02:
Sorry I haven't been able to devote enough time to this to open a PR.  My WIP branch is here: https://github.com/sammy-tri/drake/tree/jaco_example_fixes_20210202

There's still an issue with finger positions in the plan sent to the controller not being handled properly, but my arm was moving fairly smoothly with those changes.

mpatalberta, 2021-02-02:
Mr. Creasey,

// (feed forward term).
Eigen::VectorXd jaco_kp = Eigen::VectorXd::Zero(num_joints + num_fingers);
Eigen::VectorXd jaco_ki = Eigen::VectorXd::Zero(num_joints + num_fingers);
Eigen::VectorXd jaco_kd = Eigen::VectorXd::Zero(num_joints + num_fingers);
// The Jaco shows some tracking error if we're just sending open-loop
// velocities, so use a small P term to correct for that.
jaco_kp.fill(1);  <-------------------  This the line of code you only change

I had already done this. In my pull of the drake software I noticed that with this change both and the real robot seem to follow one another.
What is the relationship between the Drake and Kinova euler angles
kinova thetaX = what drake euler angle
kiinova thetaY = “
kinova thetaZ = “

I have done 2 pulls drake_kinova_driver/mpatalerta <- changed to support my arm
drake/mpatalberta  <- changed to support original and my Kinova arm

  *   Change in move_jaco to move home [should not move if you are already home],and addition command line options
  *   Change in jaco_controller  - Kp = 1 and ki,kd = 0 and additional command line options
  *   Change in jaco_simuator to send the 6 angles to put the arm home for the j2s6s300 all other worked at before , additional command line options
  *   README.md notes how to run the latest code.
Patrick Brown
Zebra Technologies,

[External Email]

mpatalberta, 2021-02-03:
Mr. Creasey,
Have you set the
KINOVAAPIUSBCOMMANDLAYER_API<http://docs.ros.org/en/indigo/api/jaco_sdk/html/Kinova_8API_8UsbCommandLayerUbuntu_8h.html#abe06f11f1aa8abc575ebd4474e559d25> int SetFrameType<http://docs.ros.org/en/indigo/api/jaco_sdk/html/Kinova_8API_8UsbCommandLayerUbuntu_8h.html#af3bd0e6780c8c6238d2de8d3da69f200>
(
int
frameType
)
Frametype can be fixed_frame
Or
Rotating Frame

This means either from the base or from the hand.
This value only takes after you have set the value and then cycle the power.
Then check the /opt/JACO/GUI/Developer on the advanced setting screen.

You can also directly set the value of the setFrametype from the developer gui.
This value need only to set once or if you have done complete system setup.
Thanks,
Patrick

mpatalberta, 2021-02-04:
I have tracked down that the random data problem comes in 2 forms.
Not the correct pid valies. Putting them all to 1 resulted both real arm and the simulator neither give a large value as random data.
So this first source is fixed.

The second source of ramdom is when you use mova_jaco.
If you use the move_jaco the first time things are great.
When you use it a second time (no coordinate change) the simulated arm in drake_visualizer move suddenly and then go to the correct position.

The above only happens to the simulated arm (kinova_simulator).

As an aside the organizers have decided that having this  j2s6s300_sphere_collision.urdf and related code will not merge as due to being in the example folder. They are complaining the example then becomes to big.

PR #15203 (2021-06-17): Fix the Jaco example to work on real hardware (not just sim)
Fixes #14593

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15203)
commit 964559e9 (2021-06-17): Fix the Jaco example to work on real hardware (not just sim) (#15203)

Fixes #14593
closed 2021-06-17 by sammy-tri